dynarithmic/twain_library #2: TWAIN DS

kamidehlvi, 2019-08-29:
hi,
I am trying to build a twain web-based application, I have downloaded sample applications from git repository the sample application part built successfully but sample DS application is not built yet gives  following error 
Error	1	error MSB6006: "cmd.exe" exited with code 3.	C:\Program Files\MSBuild\Microsoft.Cpp\v4.0\Microsoft.CppCommon.targets	149	6	TWAINDS_Sample
I have tried many solutions based on my search over web but not successful.
Please guide ...
regards

dynarithmic, 2019-08-31:
Hi,

Please note that I don't maintain the git Twain repository for the Twain data source and sample code.  That is either the Twain Working Group, or others associated with them.

Having said that, the error may be that the project is attempting to write the generated .DS file to your C:\Windows\Twain32 directory after a successful build, but does not have the rights to do so.  

Please check if the actual .DS file was created, and if the error is associated with copying that file to C:\Windows\System\Twain32 (or other system directory).  

You can look at the post build step in the project properties, and see what action is attempted to be performed.  You may have to copy the file manually (after giving yourself the proper rights to do so) to the Twain32 directory (or subdirectory of Twain32).

kamidehlvi, 2019-09-24:
> I will be closing this issue, since it isn't directly an issue with the DTWAIN library, but an issue compiling the sample source provided by the Twain Working Group.

Thank for your reply!
Regards