dynarithmic/twain_library #3: Enumerating & get/set device capabilities and image transfer mechanism in Delphi

chukkan, 2019-11-30:
Highly appreciated for a sample/demo in Delphi (Berlin or later) of enumerating the device capabilities (resolution, pixel-type, bit-depth, paper size etc.,), and get/set these capabilities, as well as a demo of different modes of image transfer mechanism.

dynarithmic, 2019-11-30:
Hello,

Basically the way to set the pixel type in Delphi is the following:

```
var
   AllPixelTypes: DTWAIN_ARRAY;
   NumPixelTypes: LONG;
   PixelValue: LONG;
   i: LONG;
   Source: DTWAIN_SOURCE;
....
  { Assume that Source has been selected }
  DTWAIN_EnumPixelTypes(Source, @AllPixelTypes);
  NumPixelTypes := DTWAIN_ArrayGetCount(AllPixelTypes);
  for i  := 0 to NumPixelTypes - 1 do
  begin
     DTWAIN_ArrayGetAtLong(AllPixelTypes, i, @PixelValue);
     { The PixelValue is one of the supported pixel types for the Source }
  end
  { Clean up }
  DTWAIN_ArrayDestroy(AllPixelTypes);

```
So basically it is a matter of sending and processing a DTWAIN_ARRAY in Delphi for many, if not all of the functions with DTWAIN_ARRAY or LPDTWAIN_ARRAY arguments,   So please extend this code to handle any function of DTWAIN that uses these argument types.

Note that for LPDTWAIN_ARRAY, we send the address of a DTWAIN_ARRAY and not the DTWAIN_ARRAY itself.  

Also, I suggest you get the latest version of DTWAIN, or at least the latest version of the Delphi bindings, since they have been updated yesterday.  The old versions had bugs (and honestly, only tested with very old versions of Delphi).

dynarithmic, 2019-11-30:
Keeping this issue open, in case you have any further issues on using DTWAIN_ARRAY functions.

dynarithmic, 2019-11-30:
Just to round this out, to get the device capabilities:

```
var
   AllCaps: DTWAIN_ARRAY;
   NumCaps: LONG;
   Cap: LONG;
   i: LONG;
   Source: DTWAIN_SOURCE;
....
  { Assume that Source has been selected }
  DTWAIN_EnumSupportedCaps(Source, @AllCaps);
  NumCaps := DTWAIN_ArrayGetCount(AllCaps);
  for i  := 0 to NumCaps - 1 do
  begin
     DTWAIN_ArrayGetAtLong(AllCaps, i, @Cap);
  end
  { Clean up }
  DTWAIN_ArrayDestroy(AllCaps);
```

Note the similarity to getting all the pixel types.

dynarithmic, 2019-12-02:
To set the pixel value:

```
var
   AllPixelTypes: DTWAIN_ARRAY;
   NumPixelTypes: LONG;
   PixelValue: LONG;
   i: LONG;
   Source: DTWAIN_SOURCE;
   FirstPixelValue: LONG;
...
  { Assume that Source has been selected }
  DTWAIN_EnumPixelTypes(Source, @AllPixelTypes);
  NumPixelTypes := DTWAIN_ArrayGetCount(AllPixelTypes);
  for i  := 0 to NumPixelTypes - 1 do
  begin
     DTWAIN_ArrayGetAtLong(AllPixelTypes, i, @PixelValue);
     { The PixelValue is one of the supported pixel types for the Source }
  end
 
  { Let's set the first available pixel type we found in the array above }
  DTWAIN_ArrayGetAtLong(AllPixelTypes, 0, @FirstPixelValue);
  DTWAIN_SetPixelType(Source, FirstPixelValue, -1, 1);
  ...
  { Clean up }
  DTWAIN_ArrayDestroy(AllPixelTypes);

```

closed 2019-12-09 by dynarithmic