dynarithmic/twain_library #4: Error On Windows 10 64 bit Acquiring from asp.net core 3.1 webapi

matteosanguin, 2019-12-16:
I'm trying to use your the wrapper you wrote "DTWAIN32U.cs" On Windows 10 64 bit from asp.net core 3.1 webapi project compiled and running in 32bit .
I don't undestand why but when i acquire 

                                        
tFileName = "d:\\scanedfile.pdf";
StringBuilder szSourceName = new StringBuilder(tFileName);
fileType = TwainAPI32.DTWAIN_PDFMULTI;

bError = TwainAPI32.DTWAIN_AcquireFile(_currentDevice, //your sample datasource 32bit
                                                 tFileName,
                                                 fileType,
                                                 (int)TwainAPI32.DTWAIN_USELONGNAME | TwainAPI32.DTWAIN_USENATIVE ,
                                                 TwainAPI32.DTWAIN_PT_DEFAULT, /* Use default */
                                                 TwainAPI32.DTWAIN_ACQUIREALL, /* Get all pages */
                                                 0, /* SHOWUI */
                                                 1,/* Close Source when UI is closed */
                                                 ref status
                                                );
i renamed your TwainAPI in TwainAPI32

i have used  for the main :

[STAThread]
public static void Main(string[] args)

i got this error :

Qt: Untested Windows version 6.2 detected!
ds: rescaling... to 1600 x 2200
QWindowsContext: OleInitialize() failed:  "COM error 0x80010106 RPC_E_CHANGED_MODE (Unknown error 0x0ffffffff80010106)"

Can you help me ?

dynarithmic, 2019-12-16:
Hi,

Does this [link](https://github.com/qmlnet/qmlnet-examples/issues/17) address the second issue (concerning the COM error)?   Let me know if it does.  

Also, the first message concerning Qt is one generated from the sample data source provided by the Twain Working Group -- you shouldn't get that error when actually using a "real" TWAIN device, or a different data source.

To remove that message, I would have to change their sample source, or figure out the Qt inner workings.   It is a very low priority item at this point.

matteosanguin, 2019-12-17:
Hi,
  thanx for the replay .
I have used the library with a kodak scanner and i have no problem with ole,qt ....

Thanx 

Matteo

closed 2019-12-17 by matteosanguin