dynarithmic/twain_library #6: Portuguese resources

mauriciofaria, 2020-01-09:
Hi.
Not an issue, but I didn't know where to put this...
I translated the language resources to Brazilian Portuguese.

[twainresourcestrings_portuguese_br.txt](https://github.com/dynarithmic/twain_library/files/4042306/twainresourcestrings_portuguese_br.txt)
 
Regards.
Maurício Faria

dynarithmic, 2020-01-10:
Hi Mauricio,

Thanks!  I will take a look at them and if ok, will add them to the next build.

dynarithmic, 2020-03-14:
Mauricio,

I added the Portuguese resource file here:

https://github.com/dynarithmic/twain_library/tree/master/additional_language_resources

The link contains instructions on using the file.  I will be closing the incident.  Thank you very much for the contribution of the language resource.

closed 2020-03-14 by dynarithmic