dynarithmic/twain_library #7: Twain library processing thread stuck sometimes

LizhuWeng, 2020-01-15:
Hi, Thanks for your great work firstly!

I found an issue that the twain_library maybe stuck sometimes, which makes the main thread has no response:

1. Happens when using DTWAIN_OpenSource API for the source like USB Camera of my laptop (Which can be get by using DTWAIN_EnumSources API)，I tested with DTWDEMO32.exe

2. It also happens when in paper jam.  already handled the message by the callback function DTWAIN_DE_PAPERJAM

3. I also had set the timeout by using DTWAIN_SetTwainTimeout API, but seems not work.

So how to handle the timeout issue properly when the Twain library block , Can you give me any advice?  Thanks!

dynarithmic, 2020-01-15:
Hi LizhuWeng,

Can you provide a log of the events when you use the USB Camera?  The DTWDEMO32.exe has a logging option in the menu.

Thanks.

LizhuWeng, 2020-01-18:
Hi, Here's the log file:
[dtwain.log](https://github.com/dynarithmic/twain_library/files/4080198/dtwain.log)
[image]

dynarithmic, 2020-01-20:
From the attached log, you're getting an error code from the Source Manager when attempting to query the ICAP_PIXELTYPE capability for the camera.  Here is a snippet from the log (at the tail end):

```
TW_MEMREF is TW_CAPABILITY:
{
    Cap=ICAP_PIXELTYPE
    ContainerType=TW_ONEVALUE
    hContainer=00000000
}

[2020-01-18 09:11:00.5428651]Output: return code from DSM=1039 (Unknown TWAIN Condition Code)

DSM_Entry(pSource=00825870H, pDest=00833BC4H, DG_CONTROL, DAT_CAPABILITY, MSG_QUERYSUPPORT, TW_MEMREF=0042B9B4H) called
Decoded pSource:
{
    Id=153
    Version Number=1.0
    Version Language=13
    Version Country=1
    Version Info=<?>
    ProtocolMajor=1
    ProtocolMinor=9
    SupportedGroups= DG_CONTROL, DG_IMAGE, DG_AUDIO
    Manufact
```

Then the log stops right there.  Note the output return code is 1039.  There is no 1039 return code defined by the current TWAIN specification that I am aware of.   Thus the problem at first glance seems to be a TWAIN Source driver issue when attempting to query your camera for the pixel types it supports.  

The driver information is here (also found in the log):
```
{
    Id=154
    Version Number=1.0
    Version Language=13
    Version Country=1
    Version Info=JULY, 2003
    ProtocolMajor=1
    ProtocolMinor=7
    SupportedGroups= DG_CONTROL, DG_IMAGE
    Manufacturer=SONIX
    Product Family=TWAIN  
    Product Name=USB Video Device
}

```

So your webcam is using the SONIX driver.  I would try and see if there is an update to this driver.  If so, install it (please get the driver from SONIX, not a no-name website), install it, and see if this corrects the issue.

dynarithmic, 2020-03-13:
Closing issue.

closed 2020-03-13 by dynarithmic