dynarithmic/twain_library #9: DTWAIN_SetOrientation not work as expected

LizhuWeng, 2020-06-07:
Hi, I encountered a problem when acquiring a file with the orientation DTWAIN_OR_LANDSCAPE  set,

It appears like just the paper size was adjusted,  but the image content was not.

[image]

LizhuWeng, 2020-06-08:
After a lot of test，it seems that there's  another CAP value to control the scanner direction （Like A4 Lanscape ）， not just by using the Orientation CAP.

Can u give me any advice? 
Thank you!

dynarithmic, 2020-06-11:
Hi,

It is impossible to know what CAP's your scanner supports without a listing of them.  You should generate a list of the caps, plus specify the scanner model you're using.

LizhuWeng, 2020-06-16:
Hi, dynarithmic.
Thanks  for reply.

I found that if not explicitly do calling DTWAIN_SetPaperSize, the result was right. but the other side still need to be set as DTWAIN_FS_A4. 

Oddly enough, If I set explicitly DTWAIN_FS_NONE, it scans the whole max area.

LizhuWeng, 2020-06-16:
SUPPORTED SIZE: 
54，54， 4，1，0， 3，5，29，2.
(Here I don't know 54 stands for what size)

ProductName: "Plustek OpticPro A380L-TWAIN
Family: "Scanner"
Manufacturer: "Plustek Inc."
VersionNo: "0.0"
CAPs: 
   ICAP_BRIGHTNESS
   ICAP_MINIMUMWIDTH
   ICAP_PIXELTYPE
   CAP_ENABLEDSUIONLY
   CAP_SERIALNUMBER
   ICAP_CONTRAST
   ICAP_BITDEPTH
   ICAP_IMAGEMERGE
   CAP_AUTOMATICSENSEMEDIUM
   ICAP_BITORDER
   ICAP_MIRROR
   ICAP_PLANARCHUNKY
   CAP_DEVICEONLINE
   CAP_DOUBLEFEEDDETECTION
   ICAP_ORIENTATION
   ICAP_AUTOMATICBORDERDETECTION
   ICAP_COMPRESSION
   CAP_CUSTOMDSDATA
   ICAP_UNDEFINEDIMAGESIZE
   CAP_EXTENDEDCAPS
   CAP_FEEDERENABLED
   ICAP_IMAGEFILEFORMAT
   CAP_INDICATORS
   ICAP_AUTOMATICROTATE
   ICAP_PHYSICALHEIGHT
   ICAP_AUTOMATICDESKEW
   ICAP_PHYSICALWIDTH
   ICAP_PIXELFLAVOR
   ICAP_SUPPORTEDEXTIMAGEINFO
   CAP_SUPPORTEDCAPS
   CAP_UICONTROLLABLE
   ICAP_UNITS
   ICAP_HIGHLIGHT
   CAP_XFERCOUNT
   ICAP_JPEGQUALITY
   ICAP_SHADOW
   ICAP_XFERMECH
   ICAP_XRESOLUTION
   ICAP_AUTOMATICCOLORENABLED
   ICAP_YRESOLUTION
   CAP_INDICATORSMODE
   ICAP_THRESHOLD
   CAP_DEVICEEVENT
   ICAP_SUPPORTEDSIZES
   ICAP_NOISEFILTER
   ICAP_GAMMA
   CAP_DUPLEX
   ICAP_LIGHTSOURCE
   CAP_PAPERDETECTABLE
   ICAP_ROTATION
   ICAP_FRAMES
   ICAP_XNATIVERESOLUTION
   ICAP_YNATIVERESOLUTION
   ICAP_MAXFRAMES
   ICAP_FLIPROTATION
   ICAP_MINIMUMHEIGHT
   ICAP_XSCALING
   CAP_SUPPORTEDDATS
   ICAP_YSCALING
   CAP_DOUBLEFEEDDETECTIONLENGTH
   ICAP_EXTIMAGEINFO
   CAP_FEEDERORDER
   CAP_FEEDERALIGNMENT
   ICAP_BARCODEDETECTIONENABLED
   ICAP_SUPPORTEDBARCODETYPES
   CAP_CUSTOMBASE + 100
   CAP_CUSTOMBASE + 111
   CAP_CUSTOMBASE + 114
   CAP_CUSTOMBASE + 115

NumCaps: 69 CustomCaps: 4 ExtendedCaps: 0

LizhuWeng, 2020-06-16:
[Here's the driver file](http://downloads.plustek.com/downloads/english/driver/OpticProA380L-V6200_3L.zip)

dynarithmic, 2020-06-16:
Thanks.  I'll take a look.  The value of 54 is for TWSS_MAXSIZE.  It basically says to acquire the entire acquisition area of the device.

dynarithmic, 2020-08-12:
I have not been able to go further with the issue, since I would basically need the scanner to duplicate.  I will close the incident, unless you need further assistance.  Also remember that the source code is available to you -- the best person to debug and to see how to address this issue is the owner of the device.  Since you have the Plustek on hand, I advise you to go through the code and see what works, suggest code changes, and see if what is suggested can be more generalized  for all scanners.

closed 2020-08-12 by dynarithmic