dynarithmic/twain_library #12: Trouble with HP ScanJet Enterprise Flow 5000 s4

olivierjc, 2020-07-17:
Hello,

I am trying to use DTWAIN with an HP ScanJet Enterprise Flow 5000 s4. It works perfectly with the demo application (VB version), but my application crashes systematically when selecting the source, without raising an exception and whatever the function used (DTWAIN_SelectSource, DTWAIN_SelectSourceByName, etc.)

I have noticed that this behavior occurs when the targeted .NET Framework version is 4.0 and above. Up to version 3.5, it works.

Since I need to use at least Framework 4.5, this situation is annoying.

Is this related to the scanner driver, in which case I just have to change it, or to DTWAIN itself?

Thank you for your reply.

dynarithmic, 2020-07-18:
Hi,

A question and observations.

1) Is this a 32-bit or 64-bit program?

2) If the VB demo program works correctly using .NET 4.5, then the issue is more than likely not DTWAIN.  DTWAIN has no .NET dependencies -- it is a non-managed DLL.  

3) You need to establish if this is a .NET issue, a marshaling / pinvoke issue, or an issue with the DTWAIN_SelectSource function itself.   You will need to see if the call to DTWAIN_SelectSource enters the DTWAIN DLL.  If it does and you are using Visual Studio, you can step into the DLL function by putting a breakpoint in the DTWAIN_SelectSource function to see if it is triggered.

dynarithmic, 2020-08-12:
Closing this issue.

closed 2020-08-12 by dynarithmic