dynarithmic/twain_library #14: Win32Exception for Lexmark Mx511de MFP in .NET Framework above 4.0 and .Net Core

frankti8899, 2020-09-14:
Hi, 

Thank you very much for this open source twain API project .

I am trying to make a C# .NET core console application for scanning from network MFP with your twain API. 

Here is my developing environment:
Windows version	: 10 ver 1809
Visual studio version	: 2019 
.Net core framework	:  3.0
Language		: C#

Dtwain version	: 5.2.01
Dtwain dll		: dtwain32u.dll
Wrapper		: DTWAIN32U.cs
Program		: 32bit

With your twain API, my application can acquire from Brother MFC-L3750CDW  and HP OfficeJet 5200. 

But it will exit with code -1073740940 (0xc0000374)  from Lexmark Mx511de  when it calls  DTWAIN_AcquireNative method.  (Lexmark MX511de seems only support Native )

/**************************************************************/
	     DTWAIN_ARRAY tempImages;
                tempImages = TwainAPI.DTWAIN_AcquireNative(SelectedSource,
                                                            TwainAPI.DTWAIN_PT_DEFAULT,
                                                            TwainAPI.DTWAIN_MAXACQUIRE,
                                                            0,
                                                            0,
                                                            ref status
                                                            );

/**************************************************************/

I  turned on  “ Enable Native code debugging” in VS2019 , found the exception thrown at ctltwmgr.cpp line 2731 .  
/**************************************************************/
 retcode = (*m_lpDSMEntry)( pOrigin, pDest, nDG, nDAT, nMSG, pData );
/**************************************************************/
VS2019 Output : 
Exception thrown at 0x755025F2 in TwainCore01.exe: Microsoft C++ exception: Win32Exception at memory location 0x02DDC328.
Critical error detected c0000374
TwainCore01.exe has triggered a breakpoint.
/**************************************************************/

Also , I tested your C# fulldemo app with Lexmark Mx511de .
Fulldemo app can work up to .Net framework 4.0.  With .Net framework above 4.0, it will exit just like my .Net Core app. 

I saw another post about HP ScanJet Enterprise Flow 5000 s4 had similar problem. 

Can you look into this issue ?

dynarithmic, 2020-09-14:
The line you pointed to:
`retcode = (*m_lpDSMEntry)( pOrigin, pDest, nDG, nDAT, nMSG, pData );`

is a call to the TWAIN Data Source Manager.  If the driver has bugs, then this is an issue with your scanner's driver.   Note that the DSM is not part of DTWAIN, but it is a component provided by the TWAIN Working Group.  

What you could try, if you haven't already, is to use the version 2.x of the TWAIN DSM, namely TWAINDSM.DLL instead of the 1.x version TWAIN_32.DLL.  You can do this by making sure that TWAINDSM.DLL is located in your application's directory, and that you direct DTWAIN to use it by calling 

`DTWAIN_SetTwainDSM(2)`

You would issue the call right after DTWAIN_SysInitialize.  The TWAINDSM.DLL you must use is the 32-bit version, and that is found [here](https://github.com/dynarithmic/twain_library/tree/master/binaries/32bit).

Please ensure that this is the DSM that is being used by either logging (DTWAIN_SetTwainLog will show you the DSM being used) or by calling DTWAIN_GetDSMFullName, or by inspecting with a tool like ProcessExplorer and seeing if TWAINDSM.DLL has been loaded.  You can test this right after you have selected the device.

What you **don't** want to use is TWAIN_32.DLL, as that will only cause the error again.

The advantage of TWAINDSM.DLL is that it is open source from the TWAIN Working Group, so if there any bugs there, then they can be discovered.  The 1.x version of the DSM is not open source, thus is very difficult to detect issues using it.

frankti8899, 2020-09-15:
Hi,

Thanks for your reply .
I added DTWAIN_SetTwainDSM(2)  and DTWAIN_SetTwainLog just after  DTWAIN_SysInitialize .   

But I still have the exception thrown at ctltwmgr.cpp line 2731 problem. 

/*********************************************/

TwainAPI.DTWAIN_SysInitialize();
int dsmVerChange = TwainAPI.DTWAIN_SetTwainDSM(2);
int setLog = TwainAPI.DTWAIN_SetTwainLog(1,"dtwain_log.txt");

/*********************************************/

Here is the record in the log file :
/*********************************************/
[2020-09-15 02:08:45.1462508]****** Log started ******
[2020-09-15 02:08:45.1834613][Enumerator Info] Creating enumerator: 07401A40
[2020-09-15 02:08:45.2453420][User] TWAIN DSM "C:\WINDOWS\system32\TWAINDSM.DLL" is found and will be used for this TWAIN session

/*********************************************/

But in the processExplorer and listDlls tools   ,  I can see both  TWAIN_32.DLL  & TWAINDSM.DLL for my application .  

/*******************************************************************************/
0x0000000010000000  0x26000   C:\WINDOWS\SysWOW64\TWAINDSM.DLL
0x00000000657d0000  0x13000   C:\WINDOWS\TWAIN_32.DLL
/*******************************************************************************/

Also TWAINDSM.DLL was loaded from SysWOW64 folder . not system32 folder .  Is there a way to force 32bit TWAINDSM.DLL to be loaded from my application’s folder ?   

The TWAIN_32.DLL &  TWAIN_32.DLL in my Windows folder/sub-folder should be installed by other applications or drivers.

dynarithmic, 2020-09-15:
Please note, I do not have the device, and it all seems to be .NET and driver related.  Thus there is little I can do from my end except to have you diagnose the problem.  Since there are literally thousands of TWAIN devices, where many of them have bugs or are not fully TWAIN compliant, the best person that can solve the issue is the programmer who has the device that is causing the issue.  

Here are a few points:

1) The error (0xc0000374)  means the heap was corrupted.  Note this is detected within either the TWAINDSM.DLL or within your driver (which has a .ds extension but is really a DLL).  I would think it isn't the TWAINDSM, as all devices use the same file.  

2) Why the DSM is loaded from SysWOW64 I am not sure.  In your SysWOW64 folder, does there already exist a TWAINDSM.DLL? The log shows that TWAINDSM.DLL has been loaded, and where it was loaded from.  Probably a Windows redirection thing.  

To control the search for the DSM:

`DTWAIN_SetDSMSearchOrder`

3) What is the exact triplet (DG / DAT / MSG) that is being issued when the error occurs?  You should either look at the log and/or record this information at the line that makes the call to the DSM.

4) What is the call stack on the call?  The call does not fail in DTWAIN.  It is failing within the TWAIN DSM (not likely) or within one of your device's executable (probably the .DS file).

5) Have you tried a very simple application, much simpler than the full demo?  An application that selects the device, acquires to a file.  Nothing fancy such as acquiring with no user interface, setting capabilities, etc.  

6) There is a TWACKER.EXE program that is available from the Twain Working Group.  I have added it [here](https://github.com/dynarithmic/twain_library/tree/master/TwainSpecification).  It is a simple C++ based application.  Does that work correctly?  

If it doesn't work correctly then the issue is definitely a driver-based issue, since TWACKER uses no code from DTWAIN whatsoever, and is an independent program..  

If it does work, then DTWAIN is making more demands on the driver to be compliant, much more than TWACKER.  So it could still be a driver issue.

7) Related to 6), do you have other TWAIN-based programs that work with your scanner?  Do **not** use the manufacturer's programs, you must use a third-party TWAIN access program.

8) Last but not least, do you have the latest drivers installed for your device?

dynarithmic, 2021-02-09:
Closing the issue.  Will reopen if there is any new information on this issue.

closed 2021-02-09 by dynarithmic