dynarithmic/twain_library #15: Save images

Krucial-kb, 2020-09-15:
How would I go about saving all the scanned images in the collection to a specified path? 

When I save the images now, it will only save 1 of the many images scanned.

dynarithmic, 2020-09-15:
There are 3 ways to do this:

1. Use [DTWAIN_SetFileAutoIncrement](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_setfileautoincrement.htm) to auto-increment the file names 

2. [DTWAIN_AcquireFileEx](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_acquirefileex.htm) if you have a set of names.

3. Trap the [DTWAIN_TN_FILENAMECHANGING notification](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_notifications_using_a_c.htm?zoom_highlightsub=notifications) using a callaback and call the [DTWAIN_SetSaveFileName](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_setsavefilename.htm) function.

dynarithmic, 2020-09-15:
Also, is documented [here](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_acquirefile.htm) near the bottom of the page.

Krucial-kb, 2020-09-15:
I’m so sorry, I’m a bit new to development, so when you call those methods...where I’m there do you tell it where you want to save the images that we’ve acquired? 

Like where do you specify the path?

dynarithmic, 2020-09-15:
Give the full path name in the second argument in the call.  

`DTWAIN_AcquireFile(Source, "SomePath\SomeFolder\SomeSubFolder\test000.bmp", etc.)`

Krucial-kb, 2020-09-15:
I love you....seriously thank you.

dynarithmic, 2020-09-15:
No problem.  Glad to help.

closed 2020-09-15 by dynarithmic