dynarithmic/twain_library #16: Replacement for FreeImage component in DTWAIN

dynarithmic, 2021-03-09:
Given the issues with the [FreeImage](https://freeimage.sourceforge.io/) imaging library, I am looking for a replacement for the FreeImage imaging code used in DTWAIN library.  FreeImage seems to be no longer supported (the last update was released 3 years ago), and it is getting to be an issue in fixing the bugs (or deficiencies) found in the existing FreeImage source code.

The last patch to FreeImage that I had to make was to fix the TIFF plugin code to save a multipage TIFF file, where each TIFF page could have differing compression schemes.  In all honesty, this should have been available in the original FreeImage plugin, since FreeImage is a high-level wrapper for [libtiff](http://www.simplesystems.org/libtiff/), which does support this.

A replacement library that is being considered is ImageMagick, or more to the point, [Magick++](https://github.com/ImageMagick/ImageMagick/tree/main/Magick%2B%2B) (or equivalent [GraphicsMagick](http://www.graphicsmagick.org/).   

There are some issues with the ImageMagick library (more DLL's will need to exist, building large multipage TIFF files are memory intensive are just the main ones).  I will attempt to address these issues when working on a POC project using this library.

Dcreeron, 2021-03-09:
Funny you should post this now. Just last night I realized my app was failing to save multi-page TIFF. I had it working previously (built LibTIFF on my own) but had dropped it when I got DTWAIN implemented. I was going to investigate tonight.

I was previously using ImageMagick/Magick++ but was able to drop that and just use Qt framework code to do what I needed.

Have you considered just building LibTIFF on your own if that's the only format you need?

dynarithmic, 2021-03-09:
In the commercial version of DTWAIN, the code directly called libtiff independently.  The goal was to consolidate all the image handling in one generic wrapper.  FreeImage was the choice for the open source version of DTWAIN, but unfortunately FreeImage seems to be out of support, and I don't have the time to find and post patches to it.

As to failing to save multipage TIFF's, FreeImage seemed to have various issues that I did patch (but didn't mention).  Probably something else has fallen through inside the TIFF handling.  TIFF's can get very complex, and I applaud the authors of libtiff keeping up with such a moving target.

Dcreeron, 2021-03-09:
Yeah, the TIFF specification/"standard" makes TWAIN look down-right easy. If I find anything tonight with FreeImage and multi-page I'll update here. Otherwise I might just go back to using my version of LibTIFF to get around this.

Dcreeron, 2021-03-11:
If you haven't looked into it yet, check out Leptonica (http://www.leptonica.org/). It supports/wraps a number of file formats including TIFF. The latest version is now using CMake and SW to build. I'm having issues getting the SW part of it to work right. It would be "lighter" than ImageMagick if you can figure out the build issues. [If you figure it out let me know. lol]

ImageMagick is easier (for me anyways) to build with the way they have it set up. But the huge amount of DLLs/LIBs, the required xml files (delegates and magic; not sure if there's an alternative to using these), mean more files to include with your "release".

dynarithmic, 2021-03-11:
Thanks for the link, I will check it out.  Right now, no final decision has been made as to the library that will be used to replace FreeImage, so all open source options (if the license is agreeable) are on the table.

Dcreeron, 2021-03-11:
I got leptonica building with VS2019, latest cmake, and latest sw last night. Ended up using these commands gleaned from this post: https://github.com/DanBloomberg/leptonica/issues/375

Not sure why the commands I was trying from their webpage didn't work. I updated it so the cmake command uses VS2019 and 32-bit.

Re-download sw (sw --self-upgrade or from the website).
Update cmake integration sw setup.

Run:

mkdir build && cd build
cmake -G "Visual Studio 16 2019" -A Win32 -DBUILD_SHARED_LIBS=1 -DSW_BUILD_SHARED_LIBS=0 ..
cmake --build .