dynarithmic/twain_library #17: Error loading library DTwain in a Java Proyect

borjafg, 2021-03-19:
I created a project using eclipse (version 2020_06) and a JDK 8 (version 241 - 32 Bits). When I tried to load the library it gives me this error:

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: C:\development\projects\ScannerTest\dlls\dtwainjni32u.dll: Can't find dependent libraries
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1934)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1850)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1122)
	at com.dynarithmic.twain.DTwainJavaAPI.DTWAIN_JavaSysInitialize(DTwainJavaAPI.java:72)
	at com.dtwain.test.DTwainJavaCapDemo.runDemo(DTwainJavaCapDemo.java:14)
	at com.dtwain.test.DTwainJavaCapDemo.main(DTwainJavaCapDemo.java:75)
```

Using a program to search the dependencies of the library I saw it needs "dtwain32.dll".

[image]

I took the library from your repository but it gives the same error.

I ran "DTWDEMO32.exe" and "DTWDEMO32U.exe" and both worked fine, so the problem must be in the java code or "dtwainjni32u.dll".

This is the java project I am using for testing to the issue. I used the source code from your repository.

[ScannerTest.zip](https://github.com/dynarithmic/twain_library/files/6172579/ScannerTest.zip)

And this is the configuration I used to launch the test.

[image]

[image]

[image]

[image]

[image]

dynarithmic, 2021-03-20:
It's been a while since I ran the Java binding, but the `dtwainjni32u.dll` has dependencies on `dtwain32u.dll`, not `dtwain32.dll`.  

[image]

You probably need to have `dtwain32u.dll` available on the system path.

borjafg, 2021-03-20:
Thanks.

Now it loads the library but it shows me another error in a cmd window when I run DTwainJavaCapDemo:

[image]

```
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10] ===>>> DTWAIN_GetVersionEx(lMajor=00DBF708, lMinor=00DBF70C, lVersionType=00DBF714, lPatchLevel=00DBF710)
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10]    ===>>> DTWAIN_GetVersionInternal(lMajor=00DBF708, lMinor=00DBF70C, lVersionType=00DBF714)
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10]    <<<=== DTWAIN_GetVersionInternal  1
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10] <<<=== DTWAIN_GetVersionEx  1
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10] ===>>> DTWAIN_GetStaticLibVersion()
[2021-03-20 17:16:10.467968] [0x0000389c] [trace]   [2021-03-20 17:16:10] <<<=== DTWAIN_GetStaticLibVersion  0
[2021-03-20 17:16:10.468969] [0x0000389c] [trace]   [2021-03-20 17:16:10] <<<=== DTWAIN_SetTwainLog  1
[2021-03-20 17:16:10.468969] [0x0000389c] [trace]   [2021-03-20 17:16:10] ===>>> DTWAIN_EnableMsgNotify(bSet=1)
[2021-03-20 17:16:10.468969] [0x0000389c] [trace]   [2021-03-20 17:16:10]    ===>>> DTWAIN_GetErrorString(lError=-1001, lpszBuffer=?, nMaxLen=1024)
[2021-03-20 17:16:10.469972] [0x0000389c] [trace]   [2021-03-20 17:16:10]    <<<=== DTWAIN_GetErrorString  0
[2021-03-20 17:16:10.469972] [0x0000389c] [trace]   [2021-03-20 17:16:10] ?

[2021-03-20 17:16:27.395720] [0x0000389c] [trace]   [2021-03-20 17:16:27]    ===>>> DTWAIN_SelectSource()
[2021-03-20 17:16:27.395720] [0x0000389c] [trace]   [2021-03-20 17:16:27]       ===>>> dynarithmic::SourceSelect(options=whichOption=1, productName=null, parentWindow=00000000, title=null, xPos=0, yPos=0, includeNames=null, excludeNames=null, nameMapping=null, options=0)
[2021-03-20 17:16:27.396717] [0x0000389c] [trace]   [2021-03-20 17:16:27]          ===>>> DTWAIN_GetErrorString(lError=-1001, lpszBuffer=, nMaxLen=1024)
[2021-03-20 17:16:27.396717] [0x0000389c] [trace]   [2021-03-20 17:16:27]          <<<=== DTWAIN_GetErrorString  0
```

[image]

[image]

[image]

dynarithmic, 2021-03-20:
Please see the [main page](https://github.com/dynarithmic/twain_library) for this error.  It is explained there.   There are text files that contains the resource strings, and DTWAIN does not find them.

borjafg, 2021-03-20:
Thanks. Now the program runs without errors.

dynarithmic, 2021-03-20:
Glad to be of help.

closed 2021-03-20 by dynarithmic

dynarithmic, 2021-03-20:
Just as a note: The Java bindings may not have all of the recently added DTWAIN functions.  The only functions available are the native ones you see in DTwainJavaAPI.java.

To add any new functions requires changes to the C++ JNI layer and the Java interface.  I haven't had the time to update the Java bindings lately, but hopefully I can get around to doing so at some point.