dynarithmic/twain_library #20: Java DTWAIN_EnumSources returns bad values in array

MichaelHurd, 2021-07-20:
Note: this error only seems to occur when using the 64-bit dll's: dtwain64(u).dll and dtwainjni64(u).dll.

When selecting a scanner using DTWAIN_SelectSource(), I get a long value that I can use to interact with the scanner. If I attempt to enumerate the sources through DTWAIN_EnumSources(), the array has a long value that does not accurately correspond to an installed scanner. This means I must select a scanner using the built-in TWAIN dialogs instead of using my program's scanner dialogs.

dynarithmic, 2021-07-21:
I will take a look and let you know what I find (with time permitting).  The JNI layer is being reworked, so it won't be surprising if this is an existing bug or oversight, given that the Java binding hasn't been updated in over a year and has not kept up with the various changes done to DTWAIN in that time.

If you have a 32-bit TWAIN driver, does the issue happen for 32-bit also?

MichaelHurd, 2021-07-21:
Thanks for the fast response! I tried the 32-bit TWAIN versions, and they were able to list my installed scanners correctly. We already have 32-bit scanner support in our application, though, and are trying to get 64-bit scanning through DTwain so we can handle very large image files.

I'm glad to hear about the JNI rework; hope it's going well. :)

dynarithmic, 2021-07-21:
I identified the problem.  The JNI layer was marshalling over 32-bit pointer values to the Java layer in 64-bit mode in the DTWAIN_EnumSources function.  

I attached an updated dtwainapi.jar, and for now just the dtwainjni64u.dll.  Please use these and check if the problem is solved.
[dtwain_java.zip](https://github.com/dynarithmic/twain_library/files/6859218/dtwain_java.zip)

MichaelHurd, 2021-07-22:
Yes, that fixed it! Thank you so much!

dynarithmic, 2021-07-23:
I updated the master branch with the Java changes.  You should be able to get all the adjusted jar files by downloading the master branch.  

In addition, the JNI source code was posted, along with the Visual Studio project file, so if you have the capabilities to do so, you can set up a recompilation of the JNI DLL's.

dynarithmic, 2021-07-23:
Closing this ticket.

closed 2021-07-23 by dynarithmic