dynarithmic/twain_library #21: Error in DTWAIN_JavaSysInitialize

MichaelHurd, 2021-07-30:
Hello again,
I think I've found an issue when initializing the DTwainJavaAPI class and loading the DLLs. The function DTWAIN_JavaSysInitialize tries to load the jni library, and then the other library, but this results in an 'UnsatisfiedLinkError' exception, with the detail "Can't find dependent libraries." If I make a call to load the other library before the jni library, it initializes correctly.

In my code, I'm attempting to use the dtwain64u.dll and dtwainjni64u.dll files.

Thanks,
Michael

dynarithmic, 2021-08-04:
Does the same issue happen for 32-bit (dtwain32u.dll and dtwainjni32u.dll)?

MichaelHurd, 2021-08-20:
Sorry, it turned out to be a problem with loading dlls and paths, not dtwain64u or dtwainjni64u.

If other people run into this issue, we figured it out by using ProcMon to monitor the attempted dll loads. For some reason, calling System.loadLibrary("dtwain64u") worked for us, but dtwain64u was unable to load TWAINDSM.dll. By moving all of the dll's and resource files into a location that ProcMon told us would always be scanned, they loaded properly.

closed 2021-08-20 by MichaelHurd