dynarithmic/twain_library #23: How can i

pengkid-rgb, 2021-11-21:


dynarithmic, 2021-11-22:
1) I am assuming you are using C or C++.

2) Are you checking the return value when calling these functions?  

3) Is this a 32-bit application or 64-bit application?  If it's a 64-bit application and all of the Twain Sources are 32-bit, you will not get any available sources.   The same thing vice-versa -- a 32-bit application cannot be used for Twain Sources that are 64-bit.

4) Did you try the DTWDEMO32U.EXE or DTWDEMO64U.exe program that is found in the binaries folder?   That example program shows a custom select source dialog, where all of the names are retrieved using DTWAIN_EnumSources and populating a dialog box with those names.

5) As to the basic way to select a source, you would use DTWAIN_SelectSourceByName() to select a TWAIN source by using the product name.  [http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_selectsourcebyname.htm](url)

pengkid-rgb, 2022-01-18:
thank you~  I tested the method you said 'DTWAIN_SelectSourceByName' and it met my needs ~

closed 2022-01-18 by pengkid-rgb