dynarithmic/twain_library #24: Slow response to Java application from Fujitsu Driver

MichaelHurd, 2021-12-23:
I get very slow results when trying to use a Fujitsu scanner from the java application. This happens both in my Java app and the sample Java DTwainDemoFrame. When debugging, it seems to happen precisely when calling the function m_api.DTWAIN_AcquireNative or m_api.DTWAIN_AcquireBuffered.

One thing I find odd, but may help diagnose the issue, is that while it's in this stuck state, the scanner instantly unsticks and starts scanning if I either lock my screen or hit Ctrl + Alt + Delete. Something about switching the Windows user mode seems to free up whatever is locking it.

The scanner driver I am using can be found here:
https://imagescanner.fujitsu.com/global/dl/setup/psip-twain64-302.html

Sorry for giving you such a tricky issue on December 23rd. Merry Christmas!

dynarithmic, 2021-12-23:
The Fujitsu scanner drivers can at times do strange things, such as lock the entire TWAIN session for any scanner to respond (Fujitsu or non-Fujitsu).   I have found that nothing short of a reboot or closing down all the Fujitsu services seems to make the scanners responsive.   Your situation does not surprise me concerning a Fujitsu scanner driver.

1) Does the scanner work for the DTWDEMO.exe or any other of the demo programs?  If so, the library itself is ok and possibly something in Java or JVM is causing the issue. 

2) What version of the JVM are you using?  8.0, 11, etc.

3) The actual Fujitsu scanner model?  

The Java framework is being redone, with numerous changes both in the JNI layer and in the Java code.  Thus the current version you are using will be obsolete as soon as the updated version is posted.  However, this will still take a few weeks before this happens.

MichaelHurd, 2021-12-23:
It does consistently work for the DTWDEMO programs.
I'm using Amazon Corretto 8.0.302 (64-bit)
I'm using an fi-7160 model.

I haven't heard about the JNI framework being redone. Do you know which version of Java the new framework will be released in? We may be able to hold out until then.

Thanks for such a fast response!

dynarithmic, 2021-12-23:
Have you tried acquiring to a file, i.e. DTWAIN_AcquireFile?  If that doesn't cause an issue, it could have something to do with the memory having to be allocated for the image(s) when issuing calls to DTWAIN_AcquireNative and DTWAIN_AcquireBuffered and the Java/JVM possibly not handling the memory correctly.  

If DTWAIN_AcquireFile works, maybe a workaround for you is to save the image to a file (BMP), and have separate Java code that processes the file, with that code being independent of DTWAIN.  

The new framework is based on Java 1.8.

MichaelHurd, 2021-12-23:
Unfortunately, it still has a tendency to freeze, even when acquiring to a file. It was worth a shot, though!

I misunderstood you earlier, I thought you meant Java itself was changing to improve upon JNI, but you meant the Java Framework for this library. That's good; it means we won't have to change out our Java version. :)
I'll will keep my fingers crossed hoping the upgrade eliminates this bug!

dynarithmic, 2021-12-23:
Can you test with another device (non-Fujitsu)?  

Also, have you tried a simple console Java application, and not a GUI one?

MichaelHurd, 2021-12-23:
Yes, the library works perfectly with our test Canon devices. Fujitsu is very common, though, so we don't want to tell people to buy different devices or lock their computers just to scan.

I haven't tried a console Java application. Since our application is client-facing, we require a GUI.

dynarithmic, 2021-12-24:
The reason why you should test a console application is to rule out whether this is an AWT/Swing issue, since the GUI runs on a different thread.   If the console application works without issues, then the issue points to threading and the GUI.

MichaelHurd, 2022-01-03:
I did get a small console app running, and unfortunately it still freezes when scanning.

dynarithmic, 2022-01-06:
I suggest you call DTWAIN_SetTwainLog, log to a file, and see where the application is frozen.  Or better yet, if you have Visual Studio, attach to the running Java process for your application from the Debug menu and choose "Break All".  Inspect the call stack to see what is idling the application.  

Since the problem does not occur in the DTWDEMO application program, the only conclusion is that Java, the JVM, or something Java related is interfering with the operation of the scanner.

dynarithmic, 2022-01-06:
As to logs, having the available logs is basically a prerequisite in diagnosing any TWAIN issues.  The DTWAIN_SetTwainLog allows the logging of all the calls made between DTWAIN and the underlying TWAIN subsystem.  

Also, you didn't mention what are the parameters being used with respect to the images being acquired.  Are these high-resolution scans, where the amount of memory for each image is relatively very high?  

But again, all of these questions are answered if logs are produced, since all of that information will be available.  As to the DTWAIN_SetTwainLog, use DTWAIN_LOG_ALL and give a file name.  The other alternative is to add a DTwainLogger instance from Java, enable it, and inspect the messages that will show up in the Java console.

MichaelHurd, 2022-01-10:
It's a pretty small image. When the program does work, it scans very quickly.
I configured the log messages, and it looks like it's freezing between these two lines:

`[2022-01-07 08:50:35]  <<<===Exiting dynarithmic::LLAcquireImage returning value of 0`
`[2022-01-07 08:55:16]  ===>>>Entering DTWAIN_IsTwainMsg(pMsg=000000F4243FA6C8)`

It definitely feels like something Java-related, but yeah, it's tricky to figure out exactly what it is.

dynarithmic, 2022-01-10:
Are you acquiring images without the scanner user interface?  If so, try acquiring with the UI enabled.

MichaelHurd, 2022-01-11:
Unfortunately, it pauses in the exact same place with or without using the UI, even if I'm scanning to a file instead of native.
I'm using the 64-bit version of the drivers, so if I'm reading the help files correctly, I'm already using the TWAINDSM.DLL.

I'll try attaching to the Java process from Visual Studio to find out precisely where this is getting stuck.

dynarithmic, 2022-01-11:
OI, If it's the 64-bit version of the Data Source Manager, that is open source from the Twain Working Group.  It's been a while since I've had to compile it, but you can get it here:

https://github.com/twain/twain-dsm/tree/master/TWAIN_DSM

It is probably newer than the version you're using, so a remote possibility is that this could fix the issue you're having.  Even it doesn't, if you can get it to compile with debugging info (Visual Studio) and replace the one you have now, then you have all the pieces, except for the Fujitsu driver code, to debug the issue.

dynarithmic, 2025-02-07:
Closing ticket, due to changes made to base DTWAIN DLL's starting with [version 5.5.5](https://github.com/dynarithmic/twain_library/releases/tag/v5.5.5).  Will reopen ticket if issues have persisted.