dynarithmic/twain_library #25: DTWAIN_IsFeederLoaded reutrns alwasy TRUE in 64-bit Delphi app

chukkan, 2022-02-11:
DTWAIN_IsFeederLoaded reutrns alwasy TRUE despite no documents loaded in ADF in 64-bit application Delphi (10.4) while 32-bit works correctly using Fujitsu fi-7260 scanner.

dynarithmic, 2022-02-11:
This could be a driver issue, since the same call works for 32-bit.  There is no difference between the 32-bit and 64-bit internal code to DTWAIN_IsFeederLoaded.

Since I do not have this scanner, you will have to turn on all logging (DTWAIN_SetTwainLog) (ignore any popups concerning capabilities not being available) and save the log to a file.  

Then inspect when DTWAIN_IsFeederLoaded gets called, the ICAP_FEEDERLOADED request to the TWAIN request will be sent.  If the request comes back that the feeder is loaded when it isn't loaded, there is nothing that DTWAIN can do about this.  This would indicate a bug in the 64-bit TWAIN driver for the scanner.

dynarithmic, 2023-03-07:
I believe this issue will be resolved once version 5.3.0.2 is released.  See the [probable duplicate issue](https://github.com/dynarithmic/twain_library-csharp_demo/issues/1).

dynarithmic, 2023-03-09:
Closing issue, and will reopen only if the latest updated DLLs (5.3.0.2) did not resolve the issue. #47