dynarithmic/twain_library #26: DTWAIN_EnumSources not working as expected

cutterman-cn, 2022-02-18:
Hi
I want to find out all devices connected, So I use the API DTWAIN_EnumSources, but turns out -1 count.
I have a Scanner connected to computer. When using official demo, it lists the device in select dialog. 
I use the example in chm file, codes below
`
DTWAIN_SOURCE MySource;
LONG nSources;
DTWAIN_ARRAY SourceArray;
TCHAR szProductName[256];
void SourceInfo() {

    if (!DTWAIN_IsTwainAvailable()) {
        printf("[ERROR] twain is not available");
        return;
    }

    if (!DTWAIN_SysInitialize()) {
        printf("[ERROR] init failed");
        return;
    }

   /* Get all of the Sources installed (note the use of the address of the DTWAIN_ARRAY) */
  DTWAIN_EnumSources(&SourceArray);
   /* Get the number of sources */
   nSources = DTWAIN_ArrayGetCount( SourceArray );
    printf("Source Count %ld", nSources);
   /* return if none found */
   if ( nSources == 0 )
   {
      DTWAIN_ArrayDestroy(SourceArray );
       return;
   }
   else /* Sources were found */
   {
       int Count;
       for (Count=0; Count < nSources; ++Count ) {
           /* Get DTWAIN_SOURCE from Array*/
          DTWAIN_ArrayGetAt(SourceArray, Count, &MySource );
          DTWAIN_GetSourceProductName(MySource, szProductName, 255 );
          printf("Source Name is %s", szProductName);
      }
      DTWAIN_ArrayDestroy( SourceArray );
  }
}

`

and the output always
`Source Count -1`

please help me, thanks.

dynarithmic, 2022-02-18:
Are you running the 32-bit version of DTWAIN or the 64-bit version?  If it's the 64-bit version, and the TWAIN drivers you have are 32-bit, then you cannot use 32-bit drivers in a 64-bit application.  The same thing vice-versa -- you cannot use 64-bit drivers in a 32-bit program.

I tested the program below, compiled as a 32-bit version, 32-bit drivers, using dtwain32.dll as the DTWAIN library, and had no issues:

```
#include <dtwain.h>
#include <stdio.h>

DTWAIN_SOURCE MySource;
LONG nSources;
DTWAIN_ARRAY SourceArray;
TCHAR szProductName[256];
void SourceInfo() {
	if (!DTWAIN_IsTwainAvailable()) {
		printf("[ERROR] twain is not available");
		return;
	}

	if (!DTWAIN_SysInitialize()) {
		printf("[ERROR] init failed");
		return;
	}

	/* Get all of the Sources installed (note the use of the address of the DTWAIN_ARRAY) */
	DTWAIN_EnumSources(&SourceArray);
	/* Get the number of sources */
	nSources = DTWAIN_ArrayGetCount(SourceArray);
	printf("Source Count %ld\n", nSources);
	/* return if none found */
	if (nSources == 0)
	{
		DTWAIN_ArrayDestroy(SourceArray);
		return;
	}
	else /* Sources were found */
	{
		int Count;
		for (Count = 0; Count < nSources; ++Count) {
			/* Get DTWAIN_SOURCE from Array*/
			DTWAIN_ArrayGetAt(SourceArray, Count, &MySource);
			DTWAIN_GetSourceProductName(MySource, szProductName, 255);
			printf("Source Name is %s\n", szProductName);
		}
		DTWAIN_ArrayDestroy(SourceArray);
	}
}

int main()
{
	SourceInfo();
}
```

The number of TWAIN drivers installed in the system I am currently on is 7:
```
Source Count 7
Source Name is EPSON GT-S50
Source Name is TWAIN2 Software Scanner
Source Name is HP CLJM277 Scan Driver TWAIN
Source Name is WIA-HP CLJM277 Scan Driver
Source Name is Plustek OpticPro A380L-TWAIN
Source Name is Plustek SmartOffice PS186-TWAIN
Source Name is Plustek PS286 Plus-TWAIN

```

Similarly, I also have 64-bit drivers, and when I build the application as a 64-bit application, the result is shown (dtwain64.dll) was used

```
Source Count 4
Source Name is Plustek OpticPro A380L-TWAIN
Source Name is TWAIN2 Software Scanner
Source Name is Plustek SmartOffice PS186-TWAIN
Source Name is Plustek PS286 Plus-TWAIN
```

So please check if you are running the correct application (32-bit, 64-bit) that matches the type of TWAIN drivers you have installed.

dynarithmic, 2022-02-18:
Also, your code does not check if any of the DTWAIN function calls return a failure code.  You should check the return codes of all of the functions.

dynarithmic, 2022-06-17:
Closing, as the issue cannot be duplicated.

closed 2022-06-17 by dynarithmic