dynarithmic/twain_library #27: How can I stop the scan ?

zongkaili, 2022-02-21:
Hello，i want to stop the scan when scanner is working， but i hava not found the corresponding API．
I return 0 when wParam is DTWAIN_TN_PAGECONTINUE in TwainCallbackProc，in fact, that means will not get the next page ，but scanner is still working. 
So, what should i do when i want to stop scanning ?  thx.

dynarithmic, 2022-02-21:
I am assuming you are using a document feeder, and transferring images using native mode (either as a raw image or acquiring the image to a file).

Even though it isn't documented, if you return 2 from DTWAIN_TN_PAGECONTINUE, does the scanner stop?  If so, then you have the autoscan mode turned on, and one way to stop the feeder is to issue a MSG_STOPFEEDER request to the scanner.  The return value of 2 will do this.

The return value of 0 *should* stop the scanner, but probably it isn't enough to stop with the particular driver you are using.

Also, it is always helpful to generate a log of all the calls being made.  The DTWAIN_SetTwainLog function will generate such a log, where all of the TWAIN traffic is logged and inspected (see the DTWDEMO.exe program for an example).

zongkaili, 2022-02-22:
I request the scanner to scan like this:
DTWAIN_SysInitialize();
DTWAIN_EnableMsgNotify(TRUE);
DTWAIN_SetCallback(TwainCallbackProc, 100);
DTWAIN_AcquireFileA(current_source, savePath.c_str(), DTWAIN_JPEG, DTWAIN_USENATIVE | DTWAIN_USENAME, 
                                   DTWAIN_PT_DEFAULT, DTWAIN_MAXACQUIRE, bShowui, TRUE, NULL);

I return 2 from DTWAIN_TN_PAGECONTINUE，the scanner does not stop. 
How to turn on the autoscan mode and issue a MSG_STOPFEEDER request to the scanner ?   I can not find the API.

dynarithmic, 2022-02-22:
There is **DTWAIN_EnableAutoScan**, or you can use [DTWAIN_SetCapValues](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_setcapvalues.htm) and specify the CAP_AUTOSCAN capability value.

dynarithmic, 2022-03-03:
You should state the make and model of your scanner, the driver version, etc.

The manufacturer of the driver is solely responsible for how the driver is designed.  Yes, there is a TWAIN specification that must be followed, but from my reading, having the ability to stop scanning and discard pages is done through the triplet:

DG_CONTROL / DAT_PENDINGXFERS / MSG_RESET

and this should have been issued when you returned 0 from DTWAIN_TN_PAGECONTINUE.  Again, the logs would show this information, and the interaction between the driver and your application.  You should verify that this is being invoked by inspecting the logs.  

Otherwise I am in the dark about exactly the interaction between the driver and your application, since I have basically no information to go by.

dynarithmic, 2022-03-03:
Also it cannot be ruled out that the manufacturer is not in accordance with the TWAIN specification, and wrote their driver with the "stop scanning" feature only available when there is a visual progress indication.  

It's impossible for me to say however, because I would need access to the driver's source code.  Very few, if any manufacturers release the source code to their drivers, let alone information that would allow a third-party to write a driver for their device.

dynarithmic, 2022-03-04:
Since the DTWAIN_TN_PAGECONTINUE is working as designed, and the issue is actually the manufacturer's driver only respecting the MSG_RESET on the pending transfers when the progress indicator is enabled, I will be closing this issue.

closed 2022-03-04 by dynarithmic

zongkaili, 2022-03-05:
Thank you for your patient answer. I got it.