dynarithmic/twain_library #28: DTWAIN_SelectSource crashes app when selecting "HP TWAIN USB 2.1 (32-32)" source

jus-ppro, 2022-02-21:
My .NET app is able to select other TWAIN sources and functions well except for this particular TWAIN source for an HP 7000 s3 scanner. After I try selecting the  "HP TWAIN USB 2.1 (32-32)" source in the Select Source dialog, my application crashes with a "The program '[24900] application.exe' has exited with code 1073741855 (0x4000001f)." error message. I tried enabling native code debugging in my project to get more detailed information, but this did not yield anything helpful. 

Interestingly, I am able to select the "HP TWAIN USB 2.1 (32-32)" TWAIN source in the [dynarithmic/twain_library-csharp_demo](https://github.com/dynarithmic/twain_library-csharp_demo) app without any issues, so I'm confused what difference in my project is causing the source selection to crash and only for this particular source.

dynarithmic, 2022-02-22:
That's strange that there are no issues with the C# demo program, but you are having an issue in your application.  

Unfortunately, I don't have this scanner to test with.

The "debug native code" is much more helpful if you have built DTWAIN from the source code (the debug version) and use **dtwain32ud.dll** instead of **dtwain32u.dll**.  Then you can actually step into the DTWAIN_SelectSource function and have basically almost end-to-end debugging (the only thing you can't debug is the HP driver itself).  The full source is in the [twain_library_source](https://github.com/dynarithmic/twain_library_source) repository (the master branch is what you would use).

Note you need Visual Studio 2015 update 3, VS 2017, or VS 2019.  If you go this route, I suggest only building the Win32_DTWAINDLL_U_D project using Win32 configuration (this is the debug, 32-bit version of the DLL) when selecting the configuration to build from the Visual Studio IDE.

There are other things you can try:

1) If you're running the program in the debugging environment, try running outside the environment.  I have had experience with manufacturer's TWAIN drivers having built anti-debugging code in their drivers.

2) Turn on logging (DTWAIN_SetTwainLog) to the highest level (the program will run slower, and you may get meaningless popup windows indicating warnings).  Then the log will show you what was going on during the Select Source call.

dynarithmic, 2022-02-22:
Also in the source repository, you can change to the development branch and get the [release libraries](https://github.com/dynarithmic/twain_library_source/blob/development/binaries/32bit/release_libraries.zip), along with the [text resources](https://github.com/dynarithmic/twain_library_source/tree/development/source/txt), and see if the unreleased version 5.2.0.9 makes any difference.  

If you plan to do this, make sure that you are using the 5.2.0.9 DLL and not version 5.2.0.8.  (You can get the version info of the running DLL by using DTWAIN_GetShortVersionString function).

jus-ppro, 2022-03-11:
Thank you! Switching to the dtwain32u.dll was helpful for troubleshooting.

closed 2022-12-05 by dynarithmic