dynarithmic/twain_library #31: How to get error details when an error occurs ?

zongkaili, 2022-04-27:
Hi,  can i get the real error details when an error which like paper jam occurs ?  
scanner :  Canon DR-6030C TWAIN  and  HUAGOSCAN G400 TWAIN

LRESULT CALLBACK TwainCallbackProc(WPARAM wParam, LPARAM lParam, LONG_PTR UserData)

For example，I set the callback as above by calling DTWAIN_SetCallback ,  I want to get the lParam  'TWAIN_ERR_FEEDERJAMMED'  when paper jam， but  wParam is 1038, lParam is -1109 and -1023  which is not the real error code.

What can I do to get the actual error code ?  please.

dynarithmic, 2022-04-28:
Have you tried calling [DTWAIN_SetDeviceNotifications](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_setdevicenotifications.htm)?  See also [DTWAIN_IsDeviceEventSupported](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_isdeviceeventsupported.htm)

Also, given the above, the DTWAIN_TN_DEVICEEVENT is the notification you should be handling.  However, this will only be successful if:
  
    a) the device supports device events, 
    b) the device properly detects a paper jam and generates the correct error message
    c) sends in the TWAIN message pump a MSG_DEVICEEVENT message that DTWAIN will process, and thus relay this in a 
        DTWAIN_TN_DEVICEEVENT notification.

Unfortunately I do not have this scanner to verify if it can actually return the error codes that you mentioned.   This is basically what you need to verify.

dynarithmic, 2022-08-15:
Closing issue

closed 2022-08-15 by dynarithmic