dynarithmic/twain_library #32: Java JNI stuck on AVISION AV830Plus scanner

ananas-hebo, 2022-05-11:
java stuck at "DTWAIN_AcquireFile",  a dialog show from the scanner driver.

[image]

```java
public class SimpleAcquire2 {
	public void SimpleTest() throws Exception {
		DTwainJavaAPI api = new DTwainJavaAPI(DTwainJavaAPI.JNIDLLName_64U);
		api.DTWAIN_JavaSysInitialize();
		api.DTWAIN_SetTwainLog(DTwainJavaAPIConstants.DTWAIN_LOG_ALL, "twain.log");

		long TwainSource = api.DTWAIN_SelectSource();
		if (TwainSource != 0) {
			int status = api.DTWAIN_AcquireFile(TwainSource, "test.png", DTwainFileType.FILETYPE_PNG,
					DTwainJavaAPIConstants.DTWAIN_USELONGNAME, DTwainJavaAPIConstants.DTWAIN_PT_DEFAULT, 1, false,
					false);
			System.out.println(status);
		}
		api.DTWAIN_JavaSysDestroy();
	}

	public static void main(String[] args) {
		SimpleAcquire2 s = new SimpleAcquire2();
		try {
			s.SimpleTest();
		} catch (Exception e) {
			e.printStackTrace();
		}
	}
}
```
[twain.log.zip](https://github.com/dynarithmic/twain_library/files/8667993/twain.log.zip)

ananas-hebo, 2022-05-11:
if remove  

```
api.DTWAIN_SetTwainLog(DTwainJavaAPIConstants.DTWAIN_LOG_ALL, "twain.log");
```
the error is gone. DTWAIN_AcquireFile return 1000.
But NO output file.

dynarithmic, 2022-05-11:
Please make sure you are looking in the correct location for "test.png".  

I would suggest to give a full path name, to verify that the file actually exists.

ananas-hebo, 2022-05-12:
@dynarithmic  thanks for your replay.

```

public class SimpleAcquire2 {
	public void SimpleTest() throws Exception {
		DTwainJavaAPI api = new DTwainJavaAPI(DTwainJavaAPI.JNIDLLName_64U);
		api.DTWAIN_JavaSysInitialize();
//		api.DTWAIN_SetTwainLog(DTwainJavaAPIConstants.DTWAIN_LOG_ALL, "twain.log");

		long TwainSource = api.DTWAIN_SelectSource();
		if (TwainSource != 0) {
			int status = api.DTWAIN_AcquireFile(TwainSource, "D:\\test.bmp", DTwainFileType.FILETYPE_BMP,
					DTwainJavaAPIConstants.DTWAIN_USELONGNAME, DTwainJavaAPIConstants.DTWAIN_PT_BW, 1, false, false);
			System.out.println(status);
		}
		api.DTWAIN_JavaSysDestroy();
	}

	public static void main(String[] args) {
		SimpleAcquire2 s = new SimpleAcquire2();
		try {
			s.SimpleTest();
		} catch (Exception e) {

			e.printStackTrace();
		}

		System.out.println("OK");
	}
}
```

output:

```
dtwainjni64u
1000
OK

```

I fix the path of the file,  but still no output file.

dynarithmic, 2022-05-12:
Please run the DTWDEMO64u.exe program to determine if this is an issue with DTWAIN itself, or a Java/JNI issue.   Right now it is impossible to tell what the issue is by looking at your Java code.

If the image can be acquired successfully using DTWAINO64u, then the underlying issue is with the Java layer in some way.  

Having said that, we are looking to overhaul the entire Java layer and utilize a new interface to the DLL through JNI.

ananas-hebo, 2022-05-16:
@dynarithmic 

i try DTWDEMO64u.exe

[image]

Acquire Native, Buffered, File is failed.
[image]

Acquire File(Source Mode)  like ok, get a tmp file:
[image]

I change my code use "DTWAIN_USELONGNAME" :

```
int status = api.DTWAIN_AcquireFile(TwainSource, "D:\\test.bmp", DTwainFileType.FILETYPE_BMP,
					DTwainJavaAPIConstants.DTWAIN_USELONGNAME + DTwainJavaAPIConstants.DTWAIN_USESOURCEMODE,
					DTwainJavaAPIConstants.DTWAIN_PT_BW, 1, false, false);
// but status return -1
```

dynarithmic, 2022-05-16:
You can generate a log from DTWDEMO by selecting logging from the main menu.  Please do that, as it seems there is something very strange about the device you are using.  

Also, using "source mode" is only available if your device itself supports the file type.  You shouldn't use it unless you are certain that your device supports acquiring to BMP files.  The list of file types can be retrieved by calling [DTWAIN_IsFileXferSupported](http://www.dynarithmic.com/onlinehelp5/dtwain/dtwain_isfilexfersupported.htm).

ananas-hebo, 2022-05-17:
@dynarithmic 

I test "TWACKER" AND [twain-samples](https://github.com/twain/twain-samples/tree/master/releases/Twain_App_sample01_02040100).

If choose output to a file are failed, but change to native is OK.
[image]

I test DTWDEMO with "native",  NO image like this: 
[image]

Use java API got a error.
``` java
public class SimpleAcquire4 {
	public void SimpleTest() throws Exception {
		DTwainJavaAPI api = new DTwainJavaAPI(DTwainJavaAPI.JNIDLLName_64U);
		api.DTWAIN_JavaSysInitialize();
		// api.DTWAIN_SetTwainLog(DTwainJavaAPIConstants.DTWAIN_LOG_ALL, "twain.log");

		long TwainSource = api.DTWAIN_SelectSource();
		if (TwainSource != 0) {
			DTwainAcquisitionArray array = api.DTWAIN_AcquireNative(TwainSource,
					DTwainJavaAPIConstants.DTWAIN_PT_DEFAULT, 1, false, false);
			System.out.println(array);
		}
		api.DTWAIN_JavaSysDestroy();
	}

	public static void main(String[] args) {
		SimpleAcquire4 s = new SimpleAcquire4();
		try {
			s.SimpleTest();
		} catch (Exception e) {
			e.printStackTrace();
		}
	}
}

dtwainjni64u
com.dynarithmic.twain.DTwainJavaAPIException: Unrecoverable error in DTWAINJavaAPI.DLL
	at com.dynarithmic.twain.DTwainJavaAPI.DTWAIN_AcquireNative(Native Method)
	at com.dtwain.test.SimpleAcquire4.SimpleTest(SimpleAcquire4.java:16)
	at com.dtwain.test.SimpleAcquire4.main(SimpleAcquire4.java:26)

```

I think dtwain has something wrong?

dynarithmic, 2022-05-17:
As mentioned previously, please turn on the logging in the DTWDEMO program.  Since I do not have your device, it is impossible to diagnose the issue until a log is presented.

ananas-hebo, 2022-05-23:
Wait a few days, I'm quarantined by COVID-19.

ananas-hebo, 2022-06-23:
I test on "Canon DR-G1100"  and "Panasonic KV-SL1077/66/56 MK2" work well.
"AVISION AV830Plus" has issues  itself.

closed 2022-06-23 by ananas-hebo