dynarithmic/twain_library #36: This application failed to start because it could not find or load the Qt platform plugin "windows" in ""

HezinTUKE, 2022-09-16:
After I select the source, I get this error **This application failed to start because it could not find or load the Qt platform plugin "windows" in ""**.
I use a dll file, I write the code in Rust

dynarithmic, 2022-09-18:
Please note that this is not a DTWAIN error -- this issue has everything to do with the data source you selected relying on the Qt framework to be properly installed.  

Since I do not use Qt, you will have to either choose a different data source that doesn't rely on Qt, or figure out why or how to get the data source you're selecting to cooperate with Qt.

HezinTUKE, 2022-09-19:
@dynarithmic I found the answer on my question - https://github.com/twain/twain-samples/issues/51#issuecomment-849417666

closed 2022-09-20 by dynarithmic