dynarithmic/twain_library #37: scanner not found using DTWAIN_SelectSource

j2t2, 2022-12-05:
One out of two scanners that I use for testing purposes (a Plustek OpticPro A360 Pro) is found as a source in the selection dialog called using DTWAIN_SelectSource(), the scanning function is working as expected. The other scanner however (a Plustek OpticSlim 1180) is not found.

There is no debug output, no error message, the result is the same as if the scanner was disconnected. I added a DTWAIN32.ini configuration and enabled DTWAIN logging to the debug console, but still got no error output. I also tried the QUERYSUPPORT=0 option.

According to the spec sheet, both of them are TWAIN compatible. Both of them correctly scan documents using an external scanning application, so the hardware and the Windows driver are working properly.

dynarithmic, 2022-12-05:
In addition, you should specify whether this is a 32-bit or 64-bit application.  If it's 32-bit, then only 32-bit TWAIN drivers can be selected.  Similarly, if it's a 64-bit application, only 64-bit drivers can be selected.

j2t2, 2022-12-05:
Thanks for your quick reply!
It's a 64-bit application using Visual Studio 2022 on Windows 10.
Unfortunately, I didn't get any error code. Before opening the selection dialog, there is no output. When selecting the scanner found, the integer error code is 0 (probably meaning no error). The other scanner doesn't show up in the selection dialog, so I can't select it to get an error message.
I enabled the logging function using DTWAIN_SetTwainLog as you described, but the only line in the logfile is "****** Log Started ******".

dynarithmic, 2023-04-19:
Closing this ticket, as the issue cannot be duplicated.

closed 2023-04-19 by dynarithmic