dynarithmic/twain_library #48: Big size of dll's

gerardcarbo, 2023-03-17:
I've seen that the size of the binary dtwain dll's are around 10-15MB in release. I think it is a quite big size for a twain wrapper. Is it normal?

dynarithmic, 2023-03-17:
The DLL includes image file libraries such as libtiff, libpng, etc., cryptography for creating encrypted PDF files, etc.  These are included in one DLL instead of spread out into many DLL's.

dynarithmic, 2023-03-17:
I experimented a bit, and was able to reduce the 32-bit Unicode DLL to a little over 9 MB (I didn't try the 64-bit DLL).  However, the following changes had to be made:

1) The compiler and runtime used is based on Visual Studio 2019 (not 2015 as it is now).  
2) The exception handling code is removed.   
3) The code to record the function call stack was removed (this is for DTWAIN_SetTwainLog).
4) Final optimization chosen  was for size, and not speed (currently it is chosen for speed).

Whether this is worth 1.5 MB savings, not sure (especially where the exception handling is turned off).  But it is something to consider in the future.

gerardcarbo, 2023-03-20:
Ok, thanks for the response. I think the savings are not worth.
On the other side I've made some fast calculations and for image+zip libraries it should take no more than 2MB. Maybe it's the crypto/pdf creation stuff, but I would say that ~8MB it's quite a lot for this.

gerardcarbo, 2023-03-21:
Ok, great! thanks.

dynarithmic, 2023-03-22:
I was able to shave off about 4MB from the size of the DLL's by removing unused imaging code from the FreeImage library, plus specifying that the DLL's be optimized for minimum size.  

I haven't fully tested them, but shortly I will put them in the development branch.  I will let you know of any further progress.