dynarithmic/twain_library #51: GetSourceProductName fillls StringBuilder with trash value

rokoduran, 2023-04-12:
Hello, I am using the C# bindings of DTWAIN. In this case, the app is built for x86, all the dll's and includes are setup properly.

As the title says, I'm unable to retrieve proper product names for sources. I nearly identically copied the code from the Full Demo that fetches the names of all enumerated sources.

        public List<string> GetConnectedSources()
        {
            List<string> sourceNames = new List<string>();
            TwainAPI.DTWAIN_SysInitialize();

            // Fill the sources
            DTWAIN_ARRAY sourceArray = IntPtr.Zero;
            TwainAPI.DTWAIN_EnumSources(ref sourceArray);

            // Get sources size
            int sourcesSize = TwainAPI.DTWAIN_ArrayGetCount(sourceArray);

            DTWAIN_SOURCE curSource = IntPtr.Zero;

            for (int i = 0; i < sourcesSize; i++)
            {
                StringBuilder sourceName = new StringBuilder(256);
                /* Get DTWAIN_SOURCE from Array*/
                TwainAPI.DTWAIN_ArrayGetSourceAt(sourceArray, i, ref curSource);
                TwainAPI.DTWAIN_GetSourceProductName(curSource, sourceName, 255);
                sourceNames.Add(sourceName.ToString());
            }
            TwainAPI.DTWAIN_ArrayDestroy(sourceArray);
            return sourceNames;
        }

But I keep getting trash values such as these: 
[image]
I have no idea what the problem is, could you help me out?

dynarithmic, 2023-04-12:
This looks like you are not using the Unicode version of the DTWAIN DLL's.  

If your application is 32-bit, you should be using **dtwain32u.dll**, for 64-bit it should be **dtwain64u.dll**.

rokoduran, 2023-04-12:
Thank you, it worked!

closed 2023-04-12 by rokoduran