dynarithmic/twain_library #54: 在有自动馈纸器的扫描仪上，如果打开了消息通知，并且在扫描过程中发生卡纸，会收到DTWAIN_TN_PAGEFAILED代码，但是消息通知回调一直不停的被调用

l585826, 2023-04-20:
我不确定我的解决方案是否合理，甚至看着有点粗暴，但是确实解决消息通知回调被不停的调用问题，我在DTWAIN_TN_PAGEFAILED回调代码中使用DTWAIN_EnableMsgNotify(0)把消息通知关掉了，在下一次进行扫描时打开。另外我使用的语言是python

dynarithmic, 2023-04-20:
Please note that the languathat should be used in the Issues section should be English.  

If I translate the text to English, I see this:

`On a scanner with an automatic document feeder, if the message notification is turned on and a paper jam occurs during the scanning process, the DTWAIN_TN_PAGEFAILED code will be received, but the message notification callback keeps being called`.

1) What is the model of the scanner?
2) Please produce a DTWAIN log of the scanning process.

l585826, 2023-04-21:
Thank you for your reply.
The scanner model is HP LaserJet M1536dnf MFP.
The Python program code is:
[TESTERR.PY.txt](https://github.com/dynarithmic/twain_library/files/11291474/TESTERR.PY.txt)

The LOG is:
[log.txt](https://github.com/dynarithmic/twain_library/files/11291461/log.txt)

dynarithmic, 2023-04-21:
As a test, in the Python program, try this:

``` 
  elif wparam == DTWAIN32.DTWAIN_TN_PAGEFAILED:
       print('DTWAIN_TN_PAGEFAILED')
       # mydll.DTWAIN_EnableMsgNotify(0)
       return 2
```
The `return 2`  will terminate the scanning session instead of retrying (that is what the return code of 1 will do).  Does this stop the failure from occuring?

It looks like the driver does not reset itself on a paper jam, and maybe the only way to reset the driver is to probably close the source using DTWAIN_CloseSource() and reopen it again.  

This is just a guess, and I would need the scanner in-house to really test this.

l585826, 2023-04-23:
`  elif wparam == DTWAIN32.DTWAIN_TN_PAGEFAILED:
       print('DTWAIN_TN_PAGEFAILED')
       # mydll.DTWAIN_EnableMsgNotify(0)
       return 0`

The `return 0` fault still occurs，but `return 2`  stopped the fault from occurring ，the DTWAIN_TN_PAGEFAILED notification received four times.
The Question 2  solved.
`images = mydll.DTWAIN_AcquireNative(
                TwainSource,                    #Source Specifies a selected TWAIN Source
                DTWAIN32.DTWAIN_PT_DEFAULT,     #PixelType Specifies the pixel type of the image
                1,                              #NumPages Specifies the number of pages to acquire   DTWAIN32.DTWAIN_MAXACQUIRE
                False,                          #bShowUI Specifies whether the Source displays the default User Interface.
                True,                           #bCloseSource Specifies whether the Source should be automatically closed when the User Interface is closed.
                byref(error_info)               #pStatus Points to a variable that will be filled in with an error status value or NULL if no error status is desired.
            )`
Change the code to:
`images = mydll.DTWAIN_AcquireNative(
                TwainSource,                    #Source Specifies a selected TWAIN Source
                DTWAIN32.DTWAIN_PT_DEFAULT,     #PixelType Specifies the pixel type of the image
                DTWAIN32.DTWAIN_MAXACQUIRE,                              #NumPages Specifies the number of pages to acquire   DTWAIN32.DTWAIN_MAXACQUIRE
                False,                          #bShowUI Specifies whether the Source displays the default User Interface.
                True,                           #bCloseSource Specifies whether the Source should be automatically closed when the User Interface is closed.
                byref(error_info)               #pStatus Points to a variable that will be filled in with an error status value or NULL if no error status is desired.
            )`,after scanning the first page, a paper jam occurred while scanning the second page,the DTWAIN_TN_PAGEFAILED notification received four times,but I don't know why.I have read the code for this library,I couldn't find the location to handle the callback return value,can you tell me?
Thank you very much again!

dynarithmic, 2023-04-23:
I made some changes to the DTWAIN_TN_PAGEFAILED notification and placed them in the development branch.   Please try the development DLL's available [here for 32-bit](https://github.com/dynarithmic/twain_library_source/tree/development/binaries/32bit) and [here for 64bit](https://github.com/dynarithmic/twain_library_source/tree/development/binaries/64bit).

As to the location of where the DTWAIN_TN_PAGEFAILED is issued it is [here](https://github.com/dynarithmic/twain_library_source/blob/31acd6d93992dfc0ed1ffdfd8064d04786f61231/source/cpp/ctltr026.cpp#L565).

dynarithmic, 2023-04-23:
Also, I highly suggest that you not attempt anything after the paper jam, as it is highly dependent on how your driver handles paper jams when something occurs.

I suggest you attempt this with the UI on, not off, and see exactly what the UI of the device does when a paper jam occurs.