dynarithmic/twain_library #55: DTwain & Windows security

johandevarwaere, 2023-04-20:
I'm building a java application which uses DTwain for scanning images with an Avision-scanner. 
Everything works fine, but I've an issue when scanning: I use the method  _DTWAIN_AcquireNative_  to start the scan. After the method is called it takes up to 30 second before the scanner actually starts scanning. 
I remarked the _Microsoft Defender Antivirus Service_ slows down the process. By trial and error I've added the _temp_ directory to the exclusion list of Microsoft Defender Antivirus. This solves the problem. More precisely it is the TWAIN.LOG file inside the temp directory causes the problem. (Excluding all .log files also solves the problem)
Do you have any idea why this happens and if it is possible to disable the log file (f.e.) /... in order not to have this problem?

Kind regards,
Johan

dynarithmic, 2023-04-20:
The TWAIN.LOG is not one produced by DTWAIN.  Not sure how to turn this off, since this seems to be one produced by the TWAIN Data Source Manager, twain_32.dll.

If possible, use the 32-bit TWAINDSM.DLL, found [here](https://github.com/dynarithmic/twain_library/tree/master/SampleTWAINDevices), and specify to use that DSM by calling [DTWAIN_SetTwainDSM](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetTwainDSM.html).

johandevarwaere, 2023-04-20:
Thanks very much, using TWAINDSM.DLL solves the problem.

dynarithmic, 2023-04-20:
Thanks.  I will close out this issue.