dynarithmic/twain_library #56: how to stop or terminate the scan

madeline0528, 2023-05-04:
I use the DTWAIN_AcquireFile api to acquire the image,but I don't want to these images,how do I stop or terminate it?

dynarithmic, 2023-05-04:
Not clear what you are asking, but if your goal is to stop the feeder, you can set up the application to get the DTWAIN_TN_PAGECONTINUE notification, and return 0 or 2 back to DTWAIN when this notification is sent. 

This is not documented fully (will need to update it), but you will see DTWAIN_TN_PAGECONTINUE explained [here](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAINNotificationCodes.html).

madeline0528, 2023-05-05:
What I mean is that,I want to stop the scanner to acquire images when the images are acquiring,which api should I use?DTWAIN_TN_PAGECONTINUE notification is dtwain triggered,not the user,is that right?

dynarithmic, 2023-05-05:
Yes, DTWAIN_TN_PAGECONTINUE is sent to the application by DTWAIN.  You need to respond to it by returning the desired value.  By default (returning a 1), it does nothing, but returning a 0 should act in stopping the scanning process.

But be aware that stopping the scanner when already set up to scan all (or a set) number of pages is highly dependent on the device as to how it will behave.  Some devices may eject all the remaining pages, some will report an error, others may stop right away and report no error.  Just be aware of the possible consequences of stopping the scanning process abruptedly.

madeline0528, 2023-05-06:
The production of scanner which I use is HuaGao,and the model of the scanner is one of these:HS-8010Pro、HS-7010Pro、HS-6010Pro、P-5010Pro、Hi-4010Pro、HS-6010Pro、HS-4010Pro、i-5010UW、Hi-5010Pro、i-6010UW、Hi-6010Pro、p-6010pro、HS-6010UW、i-6010P、G-0701W、i-6010Pro

madeline0528, 2023-05-06:
I saw the twain pdf,which write:To discard all pending images, use the DG_CONTROL / DAT_PENDINGXFERS / MSG_RESET
operation. Does the dtwain support it?

dynarithmic, 2023-05-06:
The DG_CONTROL / DAT_PENDINGXFERS / MSG_RESET will be automatically done if DTWAIN_TN_PAGECONTINUE is returned 0.

dynarithmic, 2023-05-06:
Also, if you eventually want to debug the library [source code](https://github.com/dynarithmic/twain_library_source/tree/development), basically all of the logic for DTWAIN_TN_PAGECONTINUE is found [here](https://github.com/dynarithmic/twain_library_source/blob/18a0bef044072e2977ca08e2fe0a7153f910266a/source/cpp/ctltr026.cpp#L681).  

A notification is sent out, and the return value of 0 issues the 
`ResetTransfer(MSG_RESET)`

exactly as mentioned by the TWAIN specification.

dynarithmic, 2023-05-06:
Another question:  What computer language are you using?  Is it C#, C++, etc.?

Also, I made a few changes to the libraries in the development branch.  Please test your application using the release versions found [here (32-bit)](https://github.com/dynarithmic/twain_library_source/tree/development/binaries/32bit) and [here (64-bit)](https://github.com/dynarithmic/twain_library_source/tree/development/binaries/64bit).

madeline799, 2023-05-22:
OK,thanks, and I use Javascript

madeline799, 2023-05-22:
I use the node Library ffi to call the api of dtwail.dll.But I have a question,how do I define  these variablesDG_CONTROL / DAT_PENDINGXFERS / MSG_RESET when I use DTWAIN_CallDSMProc.

dynarithmic, 2023-05-22:
The TWAIN constants are in twain.h.  

DG_CONTROL = 1
DAT_PENDINGXFERS = 5
MSG_RESET = 7

madeline799, 2023-05-22:
Thanks,and other question is when dtwain api is calling,(for example DTWAIN_AcquireFile )the other api can not be called,do you know the reason?

madeline799, 2023-05-23:
When the DTWAIN_AcquireFile is calling,I use DTWAIN_CallDSMProc to terminate the scanner,but it was not called,I have no idea about it.(The requirement is to stop the scanner if user do not want to scan.)

dynarithmic, 2023-05-23:
What verification do you have that the function was not called?  The function must have been called, and possibly you are passing in bad application and/or source identity values.  What is the return value?

This can also be verified by generating a log (DTWAIN_SetTwainLog) and seeing the call go through to the TWAIN system by looking at the function call and the sequence of steps that occur before returning.

Also, did you try what was [suggested earlier?](https://github.com/dynarithmic/twain_library/issues/56#issuecomment-1535683848) 

Finally, the scanners you mentioned are not sold in the part of the world where I am located.  The worst case scenario is to get the [source code](https://github.com/dynarithmic/twain_library_source), compile, and debug what is going on.

dynarithmic, 2024-09-26:
Closing.  Only will reopen if the issue has not been resolved.

closed 2024-09-26 by dynarithmic