dynarithmic/twain_library #59: HP Twain dialog appears after scanning, asking for the user to load the document feeder

sean-neeley, 2023-05-24:
We have a user that is using the 64bit DTWAIN library and the HP ScanJet Enterprise Flow N6600 fnw scanner.  After all of the paper is fed through the ADF, this dialog appears:

[image]

Do you know if there is a way to prevent the dialog from appearing?  Maybe an HP capability that can be turned on/off?  I know this is probably a question that is more suited for HP support, but I thought I would start here first.  Thank you for any help!

dynarithmic, 2023-05-25:
I would check if the pending transfer count is actually 0 on the last page.  If it isn't, then the issue is probably that the scanner thinks there are pages left, when the feeder is actually empty.  

You can check this by producing a log and searching for the PENDINGXFER triplet.   You should see a Count value set to 0 after all the pages have been scanned.

But having said this, just possibly the [DTWAIN_EnableIndicator](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_EnableIndicator.html) can be used to turn off the scanning progress indicator, and possibly this dialog as well.  I can't promise you this will do anything, probably it will turn off the percentage dialog (if one is showing), but maybe it will affect this dialog also.

sean-neeley, 2023-05-25:
Thanks Paul.  The feeder is definitely empty when the dialog appears.  I'll check the log and try DTWAIN_EnableIndicator.  Appreciate the ideas.

closed 2023-06-22 by sean-neeley