dynarithmic/twain_library #75: how to transfer a hexadecimal to decimal in Dtwain?

madeline799, 2023-08-09:
for example,I want to use DTWAIN_CallDSMProc,the params DG_IMAGE which hexadecimal is 0x0002L,how to transfer to decimal?Becase It has no hexadecimal in Javascript.

dynarithmic, 2023-08-09:
Hexadecimal 0x0002 is the same as decimal 2.

madeline799, 2023-08-09:
Could you show me the Java example?
Anthor,when I use the API DTWAIN_GetTwainAppID,the return value always is -1,Is the params not correct?Could you give me a hand?Is there any problem below?
const ffi = require('ffi-napi');
const ArrayType = require('ref-array-napi');
const path = require('path');
const refstruct = require('ref-struct-napi')

const version = refstruct({
    'MajorNum': 'int',
    'MinorNum': 'int',
    'Language': 'int',
    'Country': 'int',
    'Info': 'string'
})
const DTWAIN_IDENTITY = refstruct({
    'Id': 'long',
    'Version': version,
    'ProtocolMajor': 'long',
    'ProtocolMinor': 'long',
    'SupportedGroups': 'long',
    'Manufacturer': 'string',
    'ProductFamily': 'string',
    'ProductName': 'string'
});
const dtw = ffi.Library(path.join(__dirname, "dll/dtwain32.dll"), {
    "DTWAIN_GetTwainAppID": [DTWAIN_IDENTITY, []],
});

dynarithmic, 2023-08-09:
1) Sorry, but I am not familiar with the language you are using, so I cannot really help you in interfacing to a C structure using your language.

2) The Java code that shows the usage of DTWAIN_GetTwainAppID() is [here](https://github.com/dynarithmic/twain_library-java/blob/0991d052cdaf83286e7146a98fee59509fa71bf3/src/com/dynarithmic/twain/highlevel/TwainSession.java#L182).  Note that there is a TW_IDENTITY defined for Java, and that is used on the JNI side to be filled in with the TWAIN information.

dynarithmic, 2023-08-09:
Also, there is the undocumented DTWAIN_GetTwainAppIDEx function that is prototyped like this:

`TWAIN_IDENTITY DLLENTRY_DEF DTWAIN_GetTwainAppIDEx(TW_IDENTITY* pIdentity)`

The parameter passed in is a pointer to a TW_IDENTITY that is filled in by DTWAIN_GetTwainAppIDEx().  Maybe you should try to use this function instead of DTWAIN_GetTwainAppID().  

There is also a corresponding `DTWAIN_GetSourceIDEx(DTWAIN_SOURCE, TW_IDENTITY*)` that basically does the same thing, with the only difference being that you provide the source handle, and a pointer to the TW_IDENTITY to fill in.

madeline799, 2023-08-10:
Thanks a lot.I will see it.and  I have a doubt.DTWAIN_GetTwainAppID return the code -1. Do you know what is this meaning?Cause I already define the return type is Struct.

dynarithmic, 2023-08-10:
DTWAIN_GetTwainAppID returns a  `void *`, which is either 4 bytes or 8 bytes, depending on whether the application is 32-bit or 64-bit.  Since I do not know nodejs, you will need to figure out what is going on.  The return type is no different than if you called DTWAIN_SelectSource, where a 4 or 8 byte pointer is returned to you on source selection.

dynarithmic, 2023-08-25:
Closing issue, as the resolution is to declare the DTWAIN_GetTwainAppID similar to DTWAIN_SelectSource with respect to the return type.

closed 2023-08-25 by dynarithmic