dynarithmic/twain_library #76: DTWAIN_GetTwainAppID

qiaojinsong, 2023-08-10:
Hello, may I ask if it is possible to define the return value as long when using the DTWAIN_GetTwainAppID interface? And could you provide some insight into the meaning of the return codes for the DTWAIN_CallDSMProc method, such as -1 or 1?

qiaojinsong, 2023-08-10:
before： LONG DTWAIN_CallDSMProc (DTWAIN_IDENTITY AppID,DTWAIN_IDENTITY SourceId,LONG lDG,LONG lDAT,LONG lMSG,LPVOID pData );
change：LONG DTWAIN_CallDSMProc (DTWAIN_SOURCE Source,LONG lDG,LONG lDAT,LONG lMSG );

Certainly! Here's a translation of your request:

"Perhaps you can provide us with this method that directly implements DTWAIN_GetSourceID and DTWAIN_GetTwainAppID."

qiaojinsong, 2023-08-10:
i want to use LONG DTWAIN_CallDSMProc (DTWAIN_SOURCE Source,LONG lDG,LONG lDAT,LONG lMSG )，Can you provide it?

dynarithmic, 2023-08-10:
The return code is the same return code defined by the [TWAIN standard](https://github.com/dynarithmic/twain_library/blob/master/TwainSpecification/TWAIN-2.4-Specification.pdf) when you are making the triplet call.

dynarithmic, 2023-08-10:
The call to DTWAIN_CallDSMProc requires both the app ID and the source ID.  [Here is the documentation](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_CallDSMProc.html).

dynarithmic, 2023-08-10:
> "Perhaps you can provide us with this method that directly implements DTWAIN_GetSourceID and DTWAIN_GetTwainAppID."

The DTWAIN_GetSourceID() and DTWAIN_GetTwainAppID() return pointers, no different than if you called `DTWAIN_SelectSource()`, with the return value from that function also being a pointer.  

If your application can call `DTWAIN_SelectSource` succesfully, I don't see the issue with calling similar functions that return pointers.  So I cannot justify adding another function, given you have all of the pieces necessary (you still have to provide the last pointer, that will be the pointer to the data).

dynarithmic, 2023-08-11:
Your issue is similar to this one:  #75

Note that the resolution is exactly the same -- use a `void *` as the return value.

dynarithmic, 2023-08-22:
If there are no further issues pertaining to the return value, I will be closing this ticket.