dynarithmic/twain_library #82: showUI issue

mikebm, 2023-10-26:
Hello,

I've been running into a weird issue when acquiring images and enabling show UI. It is called like so:
I first initialize like so:
```
DTWAIN_SysInitialize();
DTWAIN_OpenSourcesOnSelect(DTWAIN_BOOL_TRUE);
```

Then I select a source by name.
```
DTWAIN_SelectSourceByNameW(deviceName);
```

Then each time I acquire, I run the following:
```
DTWAIN_SetBlankPageDetection(sourceDevice, 98.5,  (int)DTWAIN_BP_AUTODISCARD_ANY,  1);

DTWAIN_AcquireNativeEx(sourceDevice, TwainAPI.DTWAIN_PT_DEFAULT,
                    DTWAIN_ACQUIREALL, 1 ,  0, acquiredImagesArray, ref status)`
``

What happens is that sometimes the UI is never shown, other times it is shown, but behind every window currently open.
I am ensuring this call is happening on the UI thread (otherwise UI is never shown).

It appears that the 5th time I call the acquire function, it stops showing the UI. I am also ensuring I am not closing the source after each call.

Any help would be greatly appreciated. Thanks.

dynarithmic, 2023-10-27:
You will need to provide more information, such as the scanner model and driver version.  

The scanner's user interface is not controlled by DTWAIN at all, except only to request that the UI be shown, and DTWAIN routes messages to the UI during the scanning process.  The issue probably doesn't involve DTWAIN_SetBlankPageDetection, but you should test this by not calling this function and see what happens.

If things are just too buggy with the scanner's UI, the workaround is to not show the UI, and if possible, create your own user-interface by displaying a GUI that is under your control, and you set the different aspects of the scanner by calling the capability functions.  When "scan" is hit on your GUI, you call DTWAIN_AcquireNativeEx with the UI setting turned off.

mikebm, 2023-10-27:
Thanks for the feedback. You make a good point that I may want to just create my own UI.  I've been testing with just a simulation Twain device. I'll do some more testing a report back. Thanks again

mikebm, 2023-10-31:
Unfortunately, I can't create my own UI as some of the sensors I am working with have their own native  UI that will launch with twain. In my case I am simply trying a simulator, and the blank page doesn't have anything to do with it. I am at a loss. It works fine for a few captures then just seems to hang up my main thread. 

How my app works is that a websocket event comes in and tells the app to capture. I dispatch the select and acquire call over to the main UI thread. This app is a simple tray app that doesn't really have any UI to interact with. At some point anywhere from the 3rd to 5th call to the dispatcher, it just hangs. From what I can tell the main thread is stuck at GetMessage and my invoke call is waiting for the main thread to pick it up. I suspect something in thr acquire code is doing something to the thread that messes with it but I cannot say what.

dynarithmic, 2023-10-31:
Have you tested the TWAIN driver against the demo applications, such as DTWDEMO.EXE?  How about [Twack](https://github.com/dynarithmic/twain_library/tree/master/TwainSpecification)?  The Twack program does not use any DTWAIN code, as it was created by members of the Twain Working Group.

If the UI works correctly using those applications, then the only fix I can see is to debug the actual driver code (if that's possible).

mikebm, 2023-10-31:
> Have you tested the TWAIN driver against the demo applications, such as DTWDEMO.EXE? How about [Twack](https://github.com/dynarithmic/twain_library/tree/master/TwainSpecification)? The Twack program [...]

Both seem to work fine. I've verified my UI thread works fine without the call to this library, but eventually my UI thread stops responding to dispatcher messages with this message. 

I thought I had found a fix... by explicitly calling DTWAIN_CloseSourceUI, and calling SelectByName and CloseSource on each call. It seemed to have made it last longer (10ish+ times), but it still breaks my UI thread eventually.

dynarithmic, 2023-10-31:
ok, I was going to suggest shutting DTWAIN down completely, `DTWAIN_SysDestroy` and restarting again with `DTWAIN_SysInitialize` to see if this makes a difference.  

I don't know if this would have any effect, but maybe you can post [WM_NULL](https://learn.microsoft.com/en-us/windows/win32/winmsg/wm-null)  messages to see if the GetMessage responds.