dynarithmic/twain_library #84: DTWAIN_EnableDuplex is not working

HezinTUKE, 2023-10-30:
Hello, I have a problem with DTWAIN_EnableDuplex. I have set bEnable to true, but the source is always scanned as duplex. My code : 

```
pub fn DTWAIN_EnableDuplex(&self, lib : &Library) -> Result<bool, Box<dyn std::error::Error>> {
        unsafe {
            let duplex: libloading::Symbol<unsafe fn(*mut i32, bool) -> bool> = lib.get(b"DTWAIN_EnableDuplex").unwrap();

            Ok( duplex(self.source, false) )
        }
}
```

dynarithmic, 2023-10-30:
- I do not know Rust, so I cannot comment on the code.
- What is the scanner model?  
- You should generate a full TWAIN log.

dynarithmic, 2023-10-30:
Also, I highly suggest you test your scanner with the DTWDEMO.EXE program, or at least one of the known demo programs.  Since I don't know Rust, I have no idea what the code is doing or whether things are set up correctly.  What I do know is if the demo programs do not work for your scanner, I have something to work with.

dynarithmic, 2025-02-07:
Closing ticket due to no further issues reported.

closed 2025-02-07 by dynarithmic