dynarithmic/twain_library #85: Problem with name of file

HezinTUKE, 2023-10-30:
Hello, after scanning the source ( DTWAIN_AcquireFile ) it adds DTWAIN_SysDestroy to the file name and at the end I get the file name FILE.tifDTWAIN_SysDestroy

dynarithmic, 2023-10-30:
More than likely it is a bug in your program (maybe you are using a non-null-terminated string when a null-terminated string is expected?).  Please post the program.

dynarithmic, 2023-10-30:
[DTWAIN_AcquireFile](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_AcquireFile.html) requires a null-terminated string as the file name.  Given the issue you are having, I am getting the suspicion that the name you are giving `DTWAIN_AcquireFile` is not null-terminated.  Whatever stray characters that appear after the "good" string is being picked up, all due to the null that's missing.

If you were to produce a full TWAIN log and inspected the DTWAIN_AcquireFile call in the call stack, it will show that the string you are passing is faulty, thus not an error in DTWAIN.

dynarithmic, 2024-05-20:
Closing, as this issue is in all likelihood due to passing a string that is not null-terminated string to DTWAIN_AcquireFile.

closed 2024-05-20 by dynarithmic