dynarithmic/twain_library #88: launch  scan without any user interface

yannickModulo, 2023-12-10:
Hi,
I need to start a scan without any gui and user interaction. I have tried with modless flag but it still open a windows before starting the scanner to setup the scan. I have tried with two different models and it's the same for both.  (Windows application).
Is there a way to disable any user interface and directly launch the scan?

dynarithmic, 2023-12-10:
1) What model is the scanner?  

2) The UI is controlled by the scanner driver.  A request to not show the UI is a request -- there is no guarantee that the driver will honor the request.   

3) Did you try the DTWDEMO.exe program and see if that works with your scanner?

yannickModulo, 2023-12-10:
Hi,

thanks I have found the correct flag to disable UI. It works fine with my  all in one printer samsung. but I have an issue with the CZUR ET18. It always open a window with a previewx and with a "scanner" button to trigger the scan :( and I need to use this scanner, not the samsung.

dynarithmic, 2023-12-10:
If you used the same program to turn off the UI in the Samsung printer, but that same program does not work for the CZUR ET18, then this indicates that the manufacturer may not have this feature enabled.  

There are a few things you can do to verify if this is the case of the manufacturer not allowing scanning without the UI:

1) Call [DTWAIN_IsUIControllable](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_IsUIControllable.html) and inspect the return value.  If it is FALSE, then this means that the UI cannot be controlled, i.e. turned off.  If the return value is TRUE, then this may mean that there is a problem with the driver not reporting the correct information.

or

2) Generate a full TWAIN log ([DTWAIN_SetTwainLog](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetTwainLog.html) (DTWAIN_SetTwainLog)) and inspect the call tree when you call the DTWAIN_Acquire function.  At some point, DTWAIN will request that the UI not be set (in the DG_CONTROL / DAT_USERINTERFACE / MSG_ENABLEDS triplet with the TW_USERINTERFACE.ShowUI set to FALSE).  It may be hard to find, but doing a search in the log for `MSG_ENABLEDS` will get you to the point in the log where this is taking place.

closed 2024-09-26 by dynarithmic