dynarithmic/twain_library #90: Please could you add auto contrast and auto brightness functions support?

JanuszChmiel, 2024-01-15:
Dear elite developers,
My name is Janusz Chmiel. I do not see at all I Am visually impaired. So scanner and OCR engines are very important part of my every day life.
I have found out, that The input quality of The recognized scanned original is crucial to perform high quality OCR text result without too much mismatches and wrong words recognized.
I would like to know, how complex would be to implement Auto contrast feature.
Which could contain special mathematical formulas which would be able to automatically adjust The best contrast value while The scanning process is running. So when scanner head is moving on.
If is it technically possible while using Twain protocol. Or if such feature would require direct .sys drivers communication for The specific scanner models so this feature can not be implemented for all compatible scanning devices?
Auto contrast and The auto brightness functions Are The part of Xsane GUI app based on Xsane for Linux.
I would like to know, if auto contrast and The auto brightness could be implemented as a new property and feature to your library.
So before somebody would scan. This property, Autocontrast=true autobrightness=true would had to be set before yours scanning function would be started.
The next question is, if contrast and brightness can be automatically controlled at The same time, or to make it sense, it is only possible to use auto contrast or auto brightness, not two automatic adaptations of those scanning values at The same time.
The mathematical formula is very probably The part of Xsane app source code. Because I Am aware, that finding such routines from scratch is not trivial task.
Thank you very very much for your answer.
With The deepest possible appreciation and with The kindness regards.
Janusz Chmiel

dynarithmic, 2024-01-16:
For auto-brightness, there is the `DTWAIN_SetAutoBright` and `DTWAIN_EnableAutoBright`.  Unfortunately these functions are currently not documented (the DTWAIN documentation is lacking in describing the new functions that were added to version 5.x of DTWAIN). 

But here is an example (in C / C++):

```
DTWAIN_SOURCE source;
...
if (DTWAIN_IsAutoBrightSupported(source))
{
   DTWAIN_EnableAutoBright(TRUE);
   ...
}

```
Please note that the TWAIN driver must support the auto-bright feature (which is why the `DTWAIN_IsAutoBrightSupported` function was called).

As to contrast, there is no auto-contrast in standard TWAIN.  The only constrast function defined by standard TWAIN is available with the [DTWAIN_SetContrast](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetContrast.html) function.

If you are able to implement contrast and brightness yourself, you can always take the scanned image (as a DIB) and change it.   See [DTWAIN_SetUpdateDibProc](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetUpdateDibProc.html) for further information.  Since DTWAIN does no image manipulation (except to increase or reduce the bit depth when necessary), you will have to write the code that takes a Device Independent Bitmap (as a HANDLE), manipulate it, and return the new DIB with the changes.

JanuszChmiel, 2024-01-17:
Thank you for your elite programmers answer.
I will try The Linux operating system and Xsane GUi app if its auto contrast button bring significant OCR results improvements or no. if no, i will not be sad, that Twain do not contain this automatic contrast setup routine.
Thank you again for your explanation.

closed 2024-01-17 by JanuszChmiel