dynarithmic/twain_library #93: Driver's Native UI and Indicator Not Loading for Scanjet Pro 2000 s2 Scanner

handrenmuhammad, 2024-01-22:
Hello,

I am currently experiencing an issue with the Scanjet Pro 2000 s2 scanner when attempting to enable the driver's native UI and indicator. But the CanoScan LiDE 220 successfully loads both the UI and the indicator with the same settings. However, the Scanjet Pro 2000 s2 fails to display the expected UIs. while the Scanjet Pro 2000 s2 works fine when the UI is disabled.

what could be the issue with the Scanjet Pro 2000 s2?

dynarithmic, 2024-01-24:
Please run the DTWDEMO.exe program.  Does that program display the UI correctly?  

In general, it is impossible to know what the issue is with a particular scanner unless I have that model of scanner to test with.  

There could be multiple issues:

1) Your program could have errors causing the issue.
2) The scanner's driver is not up to date or has bugs.
3) There is some internal issue with DTWAIN, but this can only be corrected with a scanner to test with that is the same model that is giving the issue.

For 1), please check all return codes for all the DTWAIN API calls to ensure all of them return success.  If any return failure, but your application continues on as if no failure occurs, then you need to correct this.

For 2) , there is nothing DTWAIN can do about buggy TWAIN drivers.  You will need to ensure you are using the latest drivers.

For 3), If there is an issue with DTWAIN, the only way to address this is to debug the code to see where the failure occurs while using the scanner in question.  Since DTWAIN is open source, it is possible to do this, but unfortunately I do not have this scanner model to test.   

But at the minimum, you should be calling [DTWAIN_SetTwainLog](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetTwainLog.html) to generate a log file of all the TWAIN calls, and check the log for any TWAIN failures.

As to the scanner -- I have tested an HP Scanjet 7000 S3 without any issues.  The HP scanners have a similar interface, at least the 7000 does, as well as their all-in-one printer/scanner models.  If the 2000 series has the same HP interface, then I am surprised there is an issue displaying the user interface.

dynarithmic, 2024-09-26:
Closing, as this issue has not been able to be duplicated with the HP Scanjet 7000.

closed 2024-09-26 by dynarithmic