dynarithmic/twain_library #101: How to automatically close the Source UI after the scan is complete？

dongshaoyang, 2024-07-22:
When I use the DTWAIN_AcquireFile function for acquisition with bShowUI set to true, the Source UI needs to be closed manually after the acquisition is complete. How can I make it close automatically?
I didn't use the DTWAIN_SetTwainMode function to set of modal / modeless dialogs.

dynarithmic, 2024-07-22:
You can try these two things:

1) Call [DTWAIN_CloseSource](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_CloseSource.html) after the processing has been done.  The processing is done when the DTWAIN_TN_ACQUIREDONE notification is sent (you will need to set the callback, using [DTWAIN_SetCallback](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetCallback.html) or DTWAIN_SetCallback64).

or 

2) You can set the max pages parameter to 1 in the DTWAIN_AcquireFile call.  This is how the TwainSave utility accomplishes this, but granted, I don't think it's as straightforward as it should be.

I think the best bet is to use method 1).

There is no direct way right now to close the source after an acquisition.  What if the acquisition fails and the user wants to retry, or the UI just doesn't give up until the user retries?  Note that closing the UI automatically is not guaranteed to work, as there could be TWAIN UI's that require user interaction, i.e the "Cancel" or "Done" button must be pressed.

dynarithmic, 2024-07-22:
You can try [DTWAIN_SetMaxAcquisitions](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetMaxAcquisitions.html).

dynarithmic, 2024-08-16:
The suggestion of using DTWAIN_SetMaxAcquisitions should solve the issue.  Since this issue has been opened for a month, this will be closed.

closed 2024-08-16 by dynarithmic