dynarithmic/twain_library #107: Occasionally, when I call the DTWAIN_EnumSources method, there is a long delay before getting a response

dongshaoyang, 2024-08-19:
When I call the DTWAIN_EnumSources method, there is no response for a long time.

dynarithmic, 2024-08-20:
The delay can occur due to Windows having to load the DLL's (or .ds files) for each device.  This cannot be prevented.

dongshaoyang, 2024-08-21:
This is the output log file. The first few attempts didn't produce any results, only the last one was successful.
I don't know what caused it because I am not familiar with DTWAIN
[dtwain.log](https://github.com/user-attachments/files/16687334/dtwain.log)

dynarithmic, 2024-08-21:
What computer language are you using?  I see several repeated attempts of using DTWAIN_SetTwainLog.  It looks like you are calling DTWAIN_SetTwainLog in some sort of loop.  If you ran the demo DTWDEMO.exe program, do you get the same results with the speed of enumerating the sources?  

The DTWAIN_EnumSources uses TWAIN directly to go through the sources, as per the TWAIN specification using the TWAIN triplets that perform "Get the first source" and then "get the next source".  The way this works is that TWAIN itself is going to your system and finding all of the .ds extensions and calls on each .ds for basic information.  

This is what I mentioned in the previous post -- if it takes the system time to get the information, this is an issue that DTWAIN has no control over.  This is why DTWAIN caches the sources found when DTWAIN_EnumSources is called, so that subsequent calls to DTWAIN_EnumSources do not have to go through TWAIN to enumerate the sources again.

dongshaoyang, 2024-08-23:
The programming language I'm using is Go. I did call DTWAIN_SetTwainLog multiple times. As a WebSocket server, I initialize (DTWAIN_SysInitialize) and call DTWAIN_SetTwainLog each time a client establishes a connection, but these calls are made in different threads.
I ran the demo programs DTWDEMO.exe and twacker_32.exe and encountered the same issue, so it’s not a problem with DTWAIN.

dynarithmic, 2024-08-23:
It really depends on the OS and working environment.  For example, I have multiple TWAIN devices (more than 6) that take  about a second (not more than two seconds) to enumerate.   If you take those devices and installed them on another machine, the enumeration may take longer.  

If you want to know exactly what is slowing down the process, create a Go program that pauses in some way (maybe wait for a keypress) before you call DTWAIN_EnumSources.  If you have Visual Studio, you open up Visual Studio and attach to the Go process that is running.  Then open the Output Window of Visual Studio.  Then what you want to do is "unpause" the Go program and look at the Output window.  You will see all of the DLL's that are being loaded -- the DLL that shows up being "stuck" is the one that is taking the time to load.

dongshaoyang, 2024-08-23:
Thanks for the feedback.  I'll do some more testing a report back. Thank you

dynarithmic, 2024-08-23:
Since I do not use Go, if possible can you give me a small example of how you set up DTWAIN with the Go language?  I may have to make a language binding for Go (and also for Rust), so that programmers of those languages do not have issues with using DTWAIN right away,

If you can't do this, no problem.  Just that I would like to get other language examples out there, besides the ones that already exist (C/C++, C#, Python, Java, Delphi, VB).

dongshaoyang, 2024-08-24:
[Uploading dtwainDemo.zip…]()
This is a demo, very simple but it runs.

dongshaoyang, 2024-08-24:
I tried to upload this example to GitHub, but it was not successful. I am not sure why. 
So I sent you an email.

dynarithmic, 2024-08-24:
I didn't get the email.

What you can do is copy and paste a few lines of code of the declaration of the DTWAIN function and a sample line calling the code.  For example, in C#, all I would need would have been a couple of sample lines or two declaring a few of the DLL functions.

Unless in Go, you have to have a major coding setup, I should be able to get the general idea of what is required.

dongshaoyang, 2024-08-26:
[mian.zip](https://github.com/user-attachments/files/16742731/mian.zip)
ok，
A small example of how to set up DTWAIN with Go language

dynarithmic, 2024-08-26:
I opened up a new issue #109 concerning the Go language, so any updates should go there.

dongshaoyang, 2024-08-27:
Thank you very much, this is not an issue with DTWAIN, so I closed this issue.

closed 2024-08-27 by dongshaoyang