dynarithmic/twain_library #108: The response time for the DTWAIN_SelectSourceByName method is quite long

dongshaoyang, 2024-08-23:
Hello,
When I use the DTWAIN_EnumSources and DTWAIN_GetSourceProductName methods to get the product names, it always takes 10-20 seconds when I select the data source using DTWAIN_SelectSourceByName.

When I select a data source using Twack_32.exe, it is very fast. However, when I use DTWDEMO32U.exe, it takes a longer time. This happens every time I select a data source for the first time after the computer is turned on.

[dtwain_20240823.log](https://github.com/user-attachments/files/16726510/dtwain_20240823.log)

dynarithmic, 2024-08-23:
The probable reason why the call to DTWAIN_SelectSourceByName takes a long time is that by default, DTWAIN attempts to open the source and get information.  Selecting the source and opening the source are two separate operations, and by default DTWAIN_SelectSourceByName does both.

If you do not want this to happen and instead you want to manually open the source after the selection, you have two options:

1) Call [DTWAIN_OpenSourcesOnSelect](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_OpenSourcesOnSelect.html) with **0** as the argument.  Then a call to DTWAIN_SelectSourceByName will require you to also call [DTWAIN_OpenSource](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_OpenSource.html)

or

2) Call directly DTWAIN_SelectSourceByNameWithOpen where you specify the name of the file and a flag to determine if the source should also be opened:

`DTWAIN_SelectSourceByNameWithOpen(source_name, 0);`

The you will also need to call DTWAIN_OpenSource just like choice 1).  

The second choice is not documented, but it exists.  The documentation for DTWAIN hasn't been updated to include this function.

dynarithmic, 2024-08-23:
To verify where the slowness comes from, looking at your log, if you check the timestamps:

[2024-08-23 15:51:29] Thread [3604]             ===>>>Entering DTWAIN_SelectSourceByName(szProduct=Canon DR-C130 TWAIN)
```
[2024-08-23 15:51:30] Thread [3604]                ===>>>Entering DTWAIN_OpenSource(Source=03DA97E8)
...
[2024-08-23 15:51:40] Thread [3604]                <<<===Exiting DTWAIN_OpenSource  1

```

So it takes a second select the source, but it takes 9-10 seconds to open the source.   

Note that on opening the source, DTWAIN attempts to enumerate the capabilities of the Source.  This is to ensure that there are minimal (or no issues) when it is time to actually use the source for cap negotiation and image retrieval.  Twack32 does not ensure that this is the case.

dynarithmic, 2024-08-23:
Also just to add, from what I remember, Twack_32.exe opens and closes the source on each acquisition attempted (wasteful and error prone), unlike DTWAIN, which keeps the source open even after the acquisitions have been made.  

That's why the DTWAIN API requires the user to call `DTWAIN_CloseSource`, regardless of how the source was originally opened.

dongshaoyang, 2024-08-24:
Thank you very much for providing these two options. I have tried both of them, but it still takes a long time.

I believe the time is spent on "enumerating the capabilities of the source when opening it." I noticed that the DTWAIN_IsCapSupported function takes a long time.
`[2024-08-24 15:18:49] Thread [15416]                         ===>>>Entering DTWAIN_IsCapSupported(Source=00EB4538, lCapability=4098)
[2024-08-24 15:18:49] Thread [15416]                            ===>>>Entering VerifySourceHandle(DLLHandle=00E5D198, Source=00EB4538)
[2024-08-24 15:18:49] Thread [15416]                            <<<===Exiting VerifySourceHandle  00EB4538
[2024-08-24 15:19:17] Thread [15416]                         <<<===Exiting DTWAIN_IsCapSupported  1`
`[2024-08-23 15:51:31] Thread [3604]                                     ===>>>Entering DTWAIN_IsCapSupported(Source=03DA97E8, lCapability=4098)
[2024-08-23 15:51:31] Thread [3604]                                        ===>>>Entering VerifySourceHandle(DLLHandle=03D55A28, Source=03DA97E8)
[2024-08-23 15:51:31] Thread [3604]                                        <<<===Exiting VerifySourceHandle  03DA97E8
[2024-08-23 15:51:40] Thread [3604]                                     <<<===Exiting DTWAIN_IsCapSupported  1`

Is there a way to avoid enumerating the capabilities of the source when opening it, or any other way to reduce the time spent? This is a very poor experience for the users.

dongshaoyang, 2024-08-24:
Thank you very much for providing these two options. I have tried both of them, but it still takes a long time.

I believe the time is spent on "enumerating the capabilities of the source when opening it." I noticed that the DTWAIN_IsCapSupported function takes a long time.
`
[2024-08-24 15:18:49] Thread [15416]                         ===>>>Entering DTWAIN_IsCapSupported(Source=00EB4538, lCapability=4098)
[2024-08-24 15:18:49] Thread [15416]                            ===>>>Entering VerifySourceHandle(DLLHandle=00E5D198, Source=00EB4538)
[2024-08-24 15:18:49] Thread [15416]                            <<<===Exiting VerifySourceHandle  00EB4538
[2024-08-24 15:19:17] Thread [15416]                         <<<===Exiting DTWAIN_IsCapSupported  1
`
`
[2024-08-23 15:51:31] Thread [3604]                                     ===>>>Entering DTWAIN_IsCapSupported(Source=03DA97E8, lCapability=4098)
[2024-08-23 15:51:31] Thread [3604]                                        ===>>>Entering VerifySourceHandle(DLLHandle=03D55A28, Source=03DA97E8)
[2024-08-23 15:51:31] Thread [3604]                                        <<<===Exiting VerifySourceHandle  03DA97E8
[2024-08-23 15:51:40] Thread [3604]                                     <<<===Exiting DTWAIN_IsCapSupported  1
`

Is there a way to avoid enumerating the capabilities of the source when opening it, or any other way to reduce the time spent? This is a very poor experience for the users.

dongshaoyang, 2024-08-24:
Thank you very much for providing these two options. I have tried both of them, but it still takes a long time.

I believe the time is spent on "enumerating the capabilities of the source when opening it." I noticed that the DTWAIN_IsCapSupported function takes a long time.
```
[2024-08-24 15:18:49] Thread [15416]                         ===>>>Entering DTWAIN_IsCapSupported(Source=00EB4538, lCapability=4098)
[2024-08-24 15:18:49] Thread [15416]                            ===>>>Entering VerifySourceHandle(DLLHandle=00E5D198, Source=00EB4538)
[2024-08-24 15:18:49] Thread [15416]                            <<<===Exiting VerifySourceHandle  00EB4538
[2024-08-24 15:19:17] Thread [15416]                         <<<===Exiting DTWAIN_IsCapSupported  1

[2024-08-23 15:51:31] Thread [3604]                                     ===>>>Entering DTWAIN_IsCapSupported(Source=03DA97E8, lCapability=4098)
[2024-08-23 15:51:31] Thread [3604]                                        ===>>>Entering VerifySourceHandle(DLLHandle=03D55A28, Source=03DA97E8)
[2024-08-23 15:51:31] Thread [3604]                                        <<<===Exiting VerifySourceHandle  03DA97E8
[2024-08-23 15:51:40] Thread [3604]                                     <<<===Exiting DTWAIN_IsCapSupported  1
```

Is there a way to avoid enumerating the capabilities of the source when opening it, or any other way to reduce the time spent? This is a very poor experience for the users.

dynarithmic, 2024-08-24:
Honestly, I have never experienced having a source take 9 to 10 seconds to load.  

Also, the call to DTWAIN_IsCapSupported has an argument of 4098 for the capability.  Translated to hexadecimal is 0x1002.  This capability is CAP_FEEDERENABLED (you can see this in twain.h).  Thus your device is taking a long time processing this capability.

So in short, if you were to use the feeder anyway when acquiring images, the capability CAP_FEEDERENABLED would have needed to be called.  So it is either pay for the time spent on opening the source, or pay for the same time when actually acquiring an image from the feeder.

The only thing I can do is to actually change the source code and put settings in dtwain32(64).ini to not have this capability queried on opening the source.  But again, this capability will still need to be negotiated during the acquisition process.

dynarithmic, 2024-08-24:
Also, after going through your log file more thoroughly, it seems that the only real slowdown (except for when actually acquiring the image and saving to a file) is when attempting to negotiate the CAP_FEEDERENABLED capability.  

Negotiating this capability for every device I have used doesn't take more than a second.  Maybe the issue is that something is taking time in the Canon Twain driver to actually detect if the feeder is enabled (the driver needs to actually talk to the scanner hardware).  This is more of an issue with the driver than DTWAIN.  

Other than that, if you have the ability to do so, recompiling the library with coding changes may work.  The relevant lines in the DTWAIN source code are [here](https://github.com/dynarithmic/twain_library_source/blob/53a5e00d087464c3be7b6cf26f6a705b7adfbe6c/source/cpp/ctlsourceopen.cpp#L75).  Commenting out lines 75, 76, and 77 and rebuilding the source code would more than likely remove the call to query the feeder.

dynarithmic, 2024-08-26:
There are new DLL's (version 5.5.0) in the [experimental branch](https://github.com/dynarithmic/twain_library_source/tree/experimental-5.5.0/binaries) of the source repository.  See if those DLL's make any improvement on the speed of DTWAIN_OpenSource.

In addition, make sure you are using the [text resources here](https://github.com/dynarithmic/twain_library_source/tree/experimental-5.5.0/source/txt), as the INI files have a new setting that will optionally do the feeder-enabled test on opening the source.

dongshaoyang, 2024-08-27:
Okay, I'll give it a try on my end.

dongshaoyang, 2024-08-27:
I tried four different devices, but there was no change.

Here is the log using a Canon DR-C130 device.
[dtwain_canon.log](https://github.com/user-attachments/files/16759949/dtwain_canon.log)
Here is the log using a Kodak E1040 device.
[dtwain_kodak.log](https://github.com/user-attachments/files/16759955/dtwain_kodak.log)

Every time I initialize (DTWAIN_SysInitialize), the first call to DTWAIN_SelectSourceByName always takes a long time.

dynarithmic, 2024-08-27:
From looking at the logs, the devices are taking a long time to negotiate basic capabilities.  This is abnormal for devices to behave this way.  

For example, for the Canon scanner, the code is basically querying the pixel types and bit depths that the device supports, so that calls to DTWAIN_Acquire... do not fail on bad pixel types / bit depths, in addition to caching this information.

Also, the log files do not contain the timings for the TWAIN subsystem operations.  In the call to DTWAIN_SetTwainLog, you should add the following flags:

```
DTWAIN_LOG_CALLSTACK 
DTWAIN_LOG_LOWLEVELTWAIN 
DTWAIN_LOG_DECODE_TWEVENT 
DTWAIN_LOG_DECODE_TWMEMREF 

```
If the slowdown is within TWAIN32.DLL or TWAINDSM.DLL layer itself, then there is nothing that DTWAIN can do about this, as those are Windows components that control all attached TWAIN devices and how they communicate with the device hardware.

dynarithmic, 2024-08-27:
In the Canon log, there is this:

```
[2024-08-27 17:27:23.368] Thread [20812]       <<<===Exiting VerifySourceHandle returning value of 035743D8
[2024-08-27 17:27:34.224] Thread [20812]       ===>>>Entering DTWAIN_GetCapValues(Source=035743D8, lCap=257, lGetType=1, pArray=2ED1ED68)

```
Note the 11 second delay.  From looking at the code, this is exactly where DTWAIN makes a call to the TWAIN subsystem to attempt to open the source.  A complete log that includes the TWAIN subsystem interaction would break down the 11 second delay more granularly, and is the reason why you should add those flags to DTWAIN_SetTwainLog.

dongshaoyang, 2024-08-29:
I'm sorry, I'm not familiar with TWAIN. I hope you can take a look at it.
[dtwain_canon_20240828.log](https://github.com/user-attachments/files/16790382/dtwain_canon_20240828.log)

Next, I will spend some time learning about TWAIN.

dynarithmic, 2024-08-29:
I looked at the log, and it seems the slowdown occurs when DTWAIN is attempting to cache the container information.  

You should also add the flag DTWAIN_LOG_MISCELLANEOUS to the DTWAIN_SetTwainLog call and rerun the program, so that I get a list of all the capabilities that are available.  It looks like your scanner has very many capabilities that DTWAIN is going through, and I believe is a factor in the slow opening of the source.

dongshaoyang, 2024-08-29:
Okay, here is the log after adding the DTWAIN_LOG_MISCELLANEOUS flag.
[dtwain_canon_20240829.log](https://github.com/user-attachments/files/16790950/dtwain_canon_20240829.log)

dynarithmic, 2024-08-29:
Ok.  Your device has 122 capabilities, with many of them as custom capabilities.  This is probably why the open of the source is slower than most other devices I've worked with.  

I will add a couple of startup options in dtwain32.ini and dtwain64.ini that will optionally bypass the querying of the capabilities on opening of the source.   Note that this will need to be tested thoroughly to ensure that there are no unforeseen errors.  I will let you know when the libraries are ready (they will be in the experimental-5.5.0 branch of the twain_library_source repo).

dongshaoyang, 2024-08-29:
Okay, no problem. Thank you very much.

dynarithmic, 2024-08-30:
New binaries in the [experimental branch](https://github.com/dynarithmic/twain_library_source/tree/experimental-5.5.0/binaries) have been delivered.  You should try this version, plus you will need the updated [text resources](https://github.com/dynarithmic/twain_library_source/tree/experimental-5.5.0/source/txt).

The major change is the handling of DTWAIN_OpenSource and capability handling in general.  See if DTWAIN_OpenSource is faster in opening the Source.

dongshaoyang, 2024-08-31:
Yes, the source opens faster now.
[dtwain_canon_20240831.log](https://github.com/user-attachments/files/16823045/dtwain_canon_20240831.log)
[dtwain_kodak_20240831.log](https://github.com/user-attachments/files/16823047/dtwain_kodak_20240831.log)
[dtwain_other_20230831.log](https://github.com/user-attachments/files/16823048/dtwain_other_20230831.log)

Next, I will test more devices and check the functionality.

dynarithmic, 2024-08-31:
Thanks for the feedback.

Another factor is that the logging adds more time, so you can test with the logging turned off.

dynarithmic, 2024-09-26:
Closing, as version 5.5.0 and above now addresses the slow opening of TWAIN sources that contain many capabilities.

closed 2024-09-26 by dynarithmic