dynarithmic/twain_library #114: OpenCV: "Incorrect size of input array" error pop-up even with UI disabled

HelpdeskCts, 2024-09-08:
I'm working with the library in Python, making an app that should handle all errors with no pop-ups, and I'm trying to handle an error where a user pulls a card out in the middle of a scan with a Plustek MobileOffice D620. Only trying to handle this error because I see many people end up doing this.

I initialize twain with `dtwain_dll.DTWAIN_SysInitializeNoBlocking()` and set the following 
```py
self.twain_source = self.dtwain_dll.DTWAIN_SelectSourceByName(self.source_name)
self.dtwain_dll.DTWAIN_EnableDuplex(self.twain_source, True)
self.dtwain_dll.DTWAIN_EnableFeeder(self.twain_source, True)
self.dtwain_dll.DTWAIN_EnableAutoFeed(self.twain_source, False)
self.dtwain_dll.DTWAIN_EnableMsgNotify(False)
```

Then later I acquire a scan with this:
```py
result = dtwain_dll.DTWAIN_AcquireFile(source, "TEST.BMP", dtwain.DTWAIN_BMP, dtwain.DTWAIN_USELONGNAME, dtwain.DTWAIN_PT_DEFAULT, 1, False, True, 0)
``` 
with _bShowUI_ set to False to prevent any pop-ups. When I run the scan, and pull the card out while it scans. It shows this popup and all options crash my app (crashing is out of DTwain's scope most likely).
[image]
In a basic script, it returns error -1023 DTWAIN_ERR_UI_ERROR "Error in displaying Source User Interface" after pressing Ignore or Abort even though I'm not trying to show any interface whatsoever. This exact error also happens in Twack32 which is leading me to think it's something with the library. Is there something I'm missing to be able to handle this without any pop-ups?

HelpdeskCts, 2024-09-08:
And this was the minimal script I was doing to ensure no additional modules were causing issues: basically the same as the python example in the readme

```py
from ctypes import *
from dtwain32 import dtwain

class TwainScanner:
    def __init__(self):
        self.dtwain_dll = windll.LoadLibrary("./dtwain32/dtwain32u.dll")
        self.dtwain_dll.DTWAIN_SysInitializeNoBlocking()
        self.twain_source = self.dtwain_dll.DTWAIN_SelectSourceByName("Plustek MobileOffice A6 Duplex BU")
        self.dtwain_dll.DTWAIN_EnableDuplex(self.twain_source, True)
        self.dtwain_dll.DTWAIN_EnableFeeder(self.twain_source, True)
        self.dtwain_dll.DTWAIN_EnableAutoFeed(self.twain_source, False)
        self.dtwain_dll.DTWAIN_EnableMsgNotify(False)

    def get_scan(self, scan_timeout=12):
        self.dtwain_dll.DTWAIN_AcquireFile(self.twain_source, "TEST.BMP", dtwain.DTWAIN_BMP, dtwain.DTWAIN_USELONGNAME, dtwain.DTWAIN_PT_DEFAULT, 1, False, True, 0)

    def close(self):
        print(self.dtwain_dll.DTWAIN_GetLastError())
        self.dtwain_dll.DTWAIN_SysDestroy()

if __name__ == "__main__":
    twain = TwainScanner()
    twain.get_scan()
    twain.close()

```

dynarithmic, 2024-09-08:
Please generate a full DTWAIN log ([DTWAIN_SetTwainLog](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetTwainLog.html) and log all the options available using DWTAIN_LOG_ALL and give a file name).  This way, it will be more clear as to what is actually happening between the TWAIN data source manager, DTWAIN, and your application when you pull a card out.  

The -1023 DTWAIN_ERR_UI_ERROR "Error in displaying Source User Interface" is given even if you are not showing a UI.  It is given at the point where a TW_USERINTERFACE triplet is being sent to the TWAIN source manager, and that triplet contains the information on whether to start the acquisition process with or without a UI.  Since that triplet sent to TWAIN failed, the error occurs.  [See here](https://github.com/dynarithmic/twain_library_source/blob/fb72bbba0466b0192478622998ce3d329e173e88/source/cpp/ctlsourceacquire.cpp#L708) in the DTWAIN source code as to where this error occurs.  Note the comment.

Quite possibly, the Plustek driver has custom capabilities and/or [extended capabilities](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/SpecialDeviceEvents.html) that are available that detects such errors.  That is not possible for me to know this without a full TWAIN log available.  Having said this, even if Plustek has custom capabilities, unless you have information from Plustek, I will not know what those custom capabilities actually do.  

Since Twack32 has the same issue, this may be an OpenCV issue, which I have very little experience in.  Maybe you can trace into where it is asserting in the CV code and go up the call stack to see where in your code it is attempting to access the array.  From there, you could possibly do a check to ensure that the array is sized properly.

HelpdeskCts, 2024-09-08:
Thank you for the response. First log is a successful scan, second is one I have fail with the error pop-up. This time when I ran it I got -1039 DTWAIN_ERR_EXCEPTION_ERROR

[twain_log1.txt](https://github.com/user-attachments/files/16923773/twain_log1.txt)
[twain_log2.txt](https://github.com/user-attachments/files/16923774/twain_log2.txt)

If this doesn't provide any insight, I can try to reach out to Plustek about custom capabilities and update here

dynarithmic, 2024-09-08:
- Can you start the log before you call DTWAIN_OpenSource?  It is when the source is open is where I can see the capability listing.
- Are you running the latest version (5.5.0) of DTWAIN? 
- Also, it looks like your program didn't find the [twresourcestrings_english.txt](https://github.com/dynarithmic/twain_library/blob/master/text_resources/twainresourcestrings_english.txt) file, as there are some strings missing from your log.  For example:

`[2024-09-08 16:54:00] Thread [26532] =0 (TWRC_SUCCESS)`
should be:
`[2024-09-08 16:54:00] Thread [26532] Output: return code from DSM=0 (TWRC_SUCCESS)`

HelpdeskCts, 2024-09-08:
[twain_log2.txt](https://github.com/user-attachments/files/16923922/twain_log2.txt)

dynarithmic, 2024-09-09:
Looking at the capabilities, the device has the following:

```
    CAP_CUSTOMBASE + 100
    CAP_CUSTOMBASE + 111
    CAP_CUSTOMBASE + 114
    CAP_CUSTOMBASE + 115
    CAP_CUSTOMBASE + 116
    CAP_DEVICEEVENT

```
So 5 custom caps, and the CAP_DEVICEEVENT.  The CAP_DEVICEEVENT may be of interest.  See [DTWAIN_GetDeviceNotifications](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_GetDeviceNotifications.html) to get a list of notifications that the device supports.

HelpdeskCts, 2024-10-09:
I've received word with Plustek that this likely to be an issue with their driver and outside the scope of this library. I can update if I get more details, but as of now it's something they are looking into for us. Thank you.

closed 2024-10-09 by HelpdeskCts