dynarithmic/twain_library #122: Can't read TWAIN source from Array

Finn0811, 2024-09-30:
Hi, thanks a lot for providing this library.
I somehow can't read twain sources from DTWAIN_ARRAYSOURCE.

This is the code I tried with (C#):
`                // Get the total number of available sources (TWAIN devices)
                var sourceArray = TwainAPI.DTWAIN_ArrayCreate(TwainAPI.DTWAIN_ARRAYSOURCE, 0);
                int result1 = TwainAPI.DTWAIN_EnumSources(ref sourceArray);
                if (result1 == 0)
                {
                    Console.WriteLine("Error enumerating TWAIN sources.");
                    return;
                }

                int numSources = TwainAPI.DTWAIN_ArrayGetCount(sourceArray);
                if (numSources == 0)
                {
                    Console.WriteLine("No TWAIN sources found.");
                    return;
                }

                Console.WriteLine($"Found {numSources} TWAIN sources:");
                // Loop through all sources
                for (int i = 0; i < numSources; i++)
                {
                    IntPtr sourcePtr = IntPtr.Zero;

                    long arrayType = TwainAPI.DTWAIN_ArrayGetType(sourceArray);
                    Console.WriteLine("array type: " + arrayType);
                    if (TwainAPI.DTWAIN_ArrayGetAt(sourceArray, i, sourcePtr) != 0)
                    {
                        Console.WriteLine(sourcePtr);
                        // Get the product name of the source
                        StringBuilder productName = new StringBuilder(255);
                        if (TwainAPI.DTWAIN_GetSourceProductName(sourcePtr, productName, productName.Capacity) == 0)
                        {
                            Console.WriteLine($"{i + 1}. {productName.ToString()}");
                        }
                        else
                        {
                            int lastError = TwainAPI.DTWAIN_GetLastError();
                            Console.WriteLine($"Error retrieving product name for source {i + 1}. Error code: {lastError}");
                        }
                    }
                    else
                    {
                        int lastError = TwainAPI.DTWAIN_GetLastError();
                        Console.WriteLine($"Error retrieving source at index {i}. Error code: {lastError}");
                    }
                }`

This is the output:
`Found 1 TWAIN sources:
array type: 5
Error retrieving source at index 0. Error code: 0`

What am I doing wrong? :)

dynarithmic, 2024-10-01:
Use the `DTWAIN_ArrayGetSourceAt` function:

```
            DTWAIN_ARRAY pArray = System.IntPtr.Zero;
            if (TwainAPI.DTWAIN_EnumSources(ref pArray) > 0)
            {
                DTWAIN_SOURCE src = IntPtr.Zero;
                int nSources = TwainAPI.DTWAIN_ArrayGetCount(pArray);
                StringBuilder szBuf = new StringBuilder(256);
                for (int i = 0; i < nSources; ++i)
                {
                    TwainAPI.DTWAIN_ArrayGetSourceAt(pArray, i, ref src);
                    TwainAPI.DTWAIN_GetSourceProductName(src, szBuf, 256);
                    Console.WriteLine("The source name is " + szBuf);
                }
            }

```

dynarithmic, 2024-10-01:
Also, in the upcoming release of 5.5.3, the name will more closely match the other `DTWAIN_ArrayGet...At` functions, such as `DTWAIN_ArrayGetAtLong`, `DTWAIN_ArrayGetAtFloat`, etc.  It will be named `DTWAIN_ArrayGetAtSource`.

dynarithmic, 2024-10-01:
See the sample code [here](https://github.com/dynarithmic/twain_library/blob/e98f1dac138e8eb2fb1075c66d85ca020eb78d8d/demos/csharp/Examples/FullDemo/CustomSelectSource.cs#L29).

Finn0811, 2024-10-01:
Awesome, that works perfectly. Thank you! 👍 

Looks like that's the reason why I missed that function while looking at https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_ArrayGetAt.html

closed 2024-10-01 by Finn0811