dynarithmic/twain_library #126: After updating to version 5.5.3, an error occurred:DTWAIN Resource file(s) not found or corrupted: Duplicate resource ID found

dongshaoyang, 2024-10-21:
[image]

When I switch back to version 4.5.2 or 5.5.2, the issue does not occur.

dynarithmic, 2024-10-21:
1) Did you get the latest [text resources](https://github.com/dynarithmic/twain_library/tree/master/text_resources)?  
2) Do you have an older copy of twaininfo.txt in your path that is being detected by your program?

I ran the latest release 5.5.3, and do not see the issue.

dynarithmic, 2024-10-25:
The new release [5.5.4](https://github.com/dynarithmic/twain_library/releases/tag/v5.5.4) was posted.  

If your program makes multiple calls to DTWAIN_SysInitialize() and DTWAIN_SysDestroy(), there wass an issue discovered with reading the twaininfo.txt file.  That has been corrected with 5.5.4.

Not knowing how you are using DTWAIN, please try version 5.5.4 to see if you still experience the issue with twaininfo.txt.

dongshaoyang, 2024-10-28:
Yes, my program makes multiple calls to DTWAIN_SysInitialize() and DTWAIN_SysDestroy(). I tried using version 5.5.4, and the issue was resolved.

closed 2024-10-28 by dongshaoyang