dynarithmic/twain_library #130: DTWAIN_EnableDuplex set FALSE ,  DTWAIN_AcquireFileEx  just get one file but the numpages set to 3

leetb-leetb, 2024-11-21:
DTWAIN_BOOL ok = DTWAIN_SetTwainMode(DTWAIN_MODAL);
    if (!ok) {
        return false;
    }

    bool ret = DTWAIN_IsUIControllable(source);
    ret = DTWAIN_EnableIndicator(source, false);
    
    /* Also allow DTWAIN messages to be sent to our callback */
    DTWAIN_EnableMsgNotify(TRUE);
    DTWAIN_EnableTripletsNotify(TRUE);
    DTWAIN_SetCallback(TwainCallbackProc, 0);

    ret = DTWAIN_EnableDuplex(source, FALSE);  //ret return 1
    
DTWAIN_AcquireFileEx(
        hSource,
        fileNameArray,
        DTWAIN_BMP,
        DTWAIN_USELONGNAME | DTWAIN_CREATE_DIRECTORY,
        DTWAIN_PT_RGB,
        3,
        FALSE,
       FALSE,
        &status)

above is my code , user source " HUAGOSCAN G400 TWAIN";
problem is : DTWAIN_EnableDuplex set bEnbale was FALSE, DTWAIN_AcquireFileEx set numPages to 3, but just get only one file ， but DTWAIN_EnableDuplex set bEnable to true , DTWAIN_AcquireFileEx  can get 3 files.

dynarithmic, 2024-11-21:
Since DTWAIN_BMP is a single-page image type, you would be better off calling [DTWAIN_SetFileAutoIncrement](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetFileAutoIncrement.html) to set up getting the 3 files.

Also, setting the duplex unit, setting the number of pages to acquire, in addition to not showing the user-interface, is very tricky and may not give you the number of pages desired.  Many sources do not handle the page count setting correctly (even though they should).

I would recommend either showing the user-interface so the user controls the duplexing, or not show the user interface and acquire all the pages.

leetb-leetb, 2024-11-22:
after I callingDTWAIN_SetFileAutoIncrement , still cann't get the right files .  DTWAIN_EnableDuplex set to  FALSE ,  and DTWAIN_AcquireFileEx set numPages to a LONG number,  in my case , the scanner just scan a page and DTWAIN_AcquireFileEx  return Success, generate one file , the problem seems like  #81

dynarithmic, 2024-11-22:
There are many devices that do not work properly when given an exact page count to scan, whether duplex or non-duplex.   If you tried **DTWAIN_ACQUIREALL** as mentioned in #81, does the issue still occur? 

What happens if you use the scanner's user interface?  Does this work correctly?  
Does the DTWDEMO32U demo program work correctly?

Also, there are ways to further diagnose the issue at your site, since I do not have the scanner you are using:

1) Produce a full TWAIN log by calling DTWAIN_SetTwainLog().

2) Look for **PENDINGXFERS** in the page count and check the page count from TW_PENDINGXFERS.   The page count (**Count**) for each page retrieved should be non-zero.  If it is 0, then there are no more pages to scan, even if there are pages in the feeder.  If this is the case, then that is an issue with your scanner driver not reporting the correct information back to the application.

This looks like you may have to get the [source code](https://github.com/dynarithmic/twain_library_source), build it, and debug the issue.  But again, if item 2) above shows that the pending transfer count is 0 after scanning one page, there is nothing that DTWAIN can do about this, as DTWAIN behaves depending on the information the driver is sending back to DTWAIN.

dynarithmic, 2024-11-22:
Also, if you doubled the page count to **6** instead of 3, do you get all the pages?  If so, then the page count in DTWAIN_AcquireFileEx when in duplex mode should be doubled, since you are getting both the back and front side.

leetb-leetb, 2024-11-25:
Thank you for your reply. I have print all log to file , but can't find PENDINGXFERS and TW_PENDINGXFERS in logs ;
[twain.log](https://github.com/user-attachments/files/17898128/twain.log)

dynarithmic, 2024-11-25:
There are a few missing pieces:

1) The call to **DTWAIN_GetTwainMode()** shows a return value of 1.  This means that you have set the acquisition processing mode to **DTWAIN_MODELESS**, which requires that your application has to set up its own [TWAIN message loop when processing images](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/ImplementingawaitloopforModeless.html).  
The return of DTWAIN_AcquireFile() is done (successfully), and then your application has to loop to process all the TWAIN messages.  I don't see where you have set up your own loop to process messages.  So somewhere in your application, you are calling **DTWAIN_SetTwainMode(DTWAIN_MODELESS)** or **DTWAIN_SetTwainMode(1)**.

2) Did you run the DTWDEMO64 demo program, to make sure that this is not an issue with your scanner?  This will see if the issue is with your application instead of a general issue with DTWAIN.

I would suggest:

1) Do not call **DTWAIN_SetTwainMode()** at all, and instead, simply use DTWAIN_MODAL processing (the default).  If you do this, then DTWAIN does all of the message handling and looping instead of your application having to do this work.  

2) If you haven't done so, run the DTWDEMO64 program and check the results.  You may need to rebuild it if you want to test for exactly 3 pages (as the demo program always get every page in the feeder).

3) Make sure you start the log (DTWAIN_SetTwainLog) right after the call to DTWAIN_SysInitialize(), as there are function calls missing (such as the call to DTWAIN_SetTwainMode()) from the logs.

leetb-leetb, 2024-11-25:
sorry for above log, i test my program and forgot to change mode to model;
now i have call it by C++ ,  I found the callback problem :
DTWAIN_SetCallback(A::AcquireProc2, (LONG)this);  //AcquireProc2 is a static function of A

LRESULT CALLBACK A::AcquireProc2(WPARAM w, LPARAM l, LONG_PTR Data)
{
  A* pA = static_cast<A*>(Data);  
  pA->Func( );
  return TRUE;
}

static_cast<A*> can not cast to my class A object,  though I try to cast it to void *,  it still not work, sorry for that, my C++ is poor!

dynarithmic, 2024-11-25:
If your program is 64-bit, you should be calling [DTWAIN_SetCallback64](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetCallback64.html), since pointers are 64-bit, not 32-bit.   Using **DTWAIN_SetCallback**, the **this** pointer would have been truncated to 32-bit.

leetb-leetb, 2024-11-26:
Thank you for your patience. I have already resolved my issue. close the issue.

closed 2024-11-26 by leetb-leetb