dynarithmic/twain_library #131: no user interface cann't remove popup Alert window

leetb-leetb, 2024-11-22:
ret = DTWAIN_CloseSourceUI(source);
ret = DTWAIN_EnableIndicator(source, false);

/* Also allow DTWAIN messages to be sent to our callback */
DTWAIN_EnableMsgNotify(TRUE);
DTWAIN_SetCallback(TwainCallbackProc, 0);
    
DTWAIN_AcquireFileEx(
        hSource,
        fileNameArray,
        DTWAIN_BMP,
        DTWAIN_USELONGNAME | DTWAIN_CREATE_DIRECTORY,
        DTWAIN_PT_RGB,
        3,
        FALSE,
       FALSE,
        &status)

above is my code , user source " HUAGOSCAN G400 TWAIN";
my program has no UI and use dtwain with no user interface , TwainCallbackProc cannot recieve the notifications like:  "no papers" , "the source invalid"  , "paper jam", it always popup a alert  with mentioned  notifications , is there going wrong?

dynarithmic, 2024-11-22:
Any alert boxes are generated by the driver, not DTWAIN.  DTWAIN has no control over this, unless there are custom capabilities that your scanner driver has that are available to turn off these notifications.

Also, detection of "no paper" can only be done by your application calling DTWAIN_IsFeederLoaded().  Note that your TWAIN driver has to support the capabilities that tests if a feeder is able to detect whether paper is loaded [DTWAIN_IsFeederSensitive()](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_IsFeederSensitive.html).  

You should list all the TWAIN capabilities of your device, since I have no idea what they are.  Depending on the capability listing, possibly you can set the correct ones to turn off notifications and/or detect paper jams.

leetb-leetb, 2024-11-25:
Thank you for your reply.
I have print all the TWAIN capabilities of my device. it seems to capality to turn off notifications and detect paper jams.
[G400.txt](https://github.com/user-attachments/files/17896716/G400.txt)

dynarithmic, 2024-11-25:
1) I see that your device has CAP_FEEDERLOADED.  For the issue with the "no paper", you can write a loop to wait until CAP_FEEDERLOADED returns true.  Once it returns true, then call DTWAIN_AcquireFile(Ex).  This is exactly how the WaitForFeederLoaded demo program works.  The only difference is that WaitForFeederLoaded has a timer set so that if no paper is detected in a certain time frame, then DTWAIN_Acquire...() is not called, and the acquisition isn't done. 

Something similar to this:

```
DTWAIN_SOURCE source;
//....
while (/* time has not expired */)
{
    if ( DTWAIN_IsFeederLoaded(source) )
      DTWAIN_AcquireFile(...);
}

```
2) You have many capabilities that are custom (CAP_CUSTOMBASE + x).  I do not know what these capabilities are or what they accomplish, since the manufacturer of your device has this information.  Maybe one or more of these capabilities can control the notifications.   

Starting with version **5.5.0**, the function [DTWAIN_TestGetCap](https://github.com/dynarithmic/twain_library/blob/d1f50b1ac63a53f6aa38edb7bb72b7bc8bc9a3c4/updates/updates.txt#L164) is available to see what container types and data types will cause a capability to return information (whether the information is valid or not, you will have to determine this by inspection).

leetb-leetb, 2024-11-25:
thanks guys !,  i close the issue.

closed 2024-11-25 by leetb-leetb