dynarithmic/twain_library #132: occur crash after call DTWAIN_SysDestroy() in C++ program

leetb-leetb, 2024-11-27:
when open an device which no exist, but have driver installed , popup alert window tips that: "not found device" , clicked "ok" button and finish program , occur crash after call DTWAIN_SysDestroy();  DTWAIN_EnableMsgNotify(TRUE) has been called and callback return 1 for every notifications.
 0x00007FF9E8D94613 (CGEHDll64.dll) (example.exe 中) exist an unprocessed exception: 0xC000041D: During the user callback, an unprocessed exception was encountered.
is there something wrong ?

dynarithmic, 2024-11-27:
1) Please note that it is not possible to diagnose this error without a simple sample program that shows the issue.  At the very least you should *always* generate a log file, so that you (and myself) can see exactly what is happening on each call.

2) `CGEHDll64.dll` -- The exception occurs here.  What application does this DLL belong to? Is this part of your TWAIN scanner's software?

3)  Did `DTWAIN_SelectSource()` return a null source handle, or did it return a valid Source?   If the source was valid, then your driver is reporting that the source was open, but the device does not exist (this is inconsistent).  If this happens, have you tried calling `DTWAIN_CloseSource()`, to make sure the source is closed?

4) `DTWAIN_SysDestroy()` attempts to close any open sources.  If I were to guess, the error comes from your TWAIN driver not being able to be closed when the device doesn't exist, thus the exception in `CGEHDll64.dll`.  If this is the case, there is nothing that DTWAIN can do about this, as it is relying on your driver to behave correctly when requested to be closed.

dynarithmic, 2024-11-28:
1) What version of DTWAIN are you using?  Are you using the latest (5.5.4) version?

The logs you are showing doesn't show the DTWAIN version.  The version information should be near the top, similar to this:
`Thread [60568] Dynarithmic TWAIN Library, Version 5.5`

2) There is an error attempting to close the data source manager.  I believe the driver corrupts the DSM in some way:
```
[2024-11-28 11:04:40.127] Thread [14516] Input: 
DSM_Entry(pSource=000002880D78EFB8H, pDest=0000000000000000H, DG_CONTROL, DAT_PARENT, MSG_CLOSEDSM, TW_MEMREF=000002880D78EF90H) called
Decoded pSource:
{
....
}

No information for pDest

TW_MEMREF is handle to window (HWND):
{
    HWND=000000000014087C
    Screen Pos.= 0,0,136,39
}

[2024-11-28 11:04:40.127] Thread [14516] Output: return code from DSM=1 (TWRC_FAILURE)

```
Closing the DSM should not result in an error if it was previously opened.  The TWAINDSM.DLL file is the component that is called on to be closed.  My conclusion is that your scanner driver is faulty.  Since you mentioned that the real exception occurs in `CGEHDll64.dll`, this leads me to believe it is the driver that is not behaving correctly if the device doesn't exist.

The call to `DTWAIN_OpenSource()` returns 0, which is expected if the source cannot be opened after being selected.  So up to this point, the application seems ok.  But if your driver leaves some sort of artifacts around for the TWAINDSM.DLL to believe the source is still "alive", then I am not surprised that a crash occurs if the TWAINDSM.DLL has trouble closing itself.

Also, since the DTWAIN_SysDestroy() did not complete, you probably won't be able to call `DTWAIN_SysInitialize()` later on in your application (if you want to do so).

If you are interested, the point in the DTWAIN source code where the DSM is attempted to be closed is [here](https://github.com/dynarithmic/twain_library_source/blob/da69d540dac2bfc1054eec25d1c25184dfe0dd71/source/cpp/ctldtwaininit.cpp#L1612).  DTWAIN has a global `try/catch` for exceptions that may occur when closing the DSM, but it looks like the application is corrupted to the point where the crash occurs anyway.

The only thing I can do is in a future version of DTWAIN is to try and not log anything, but simply write to the debug log window that the DSM could not be closed.

leetb-leetb, 2024-11-28:
used DTWAIN_GetVersionString print version was : Dynarithmic TWAIN Library, Version 5.5 -  (64-bit) Version (Patch Level 4) Internal Build Number: 001
Shared Library path : xxx\build\Debug\DTWAIN64.dll
Using Resource file (twaininfo.txt) version: 5.5.4
I had solved the crash problem, just remove DTWAIN_SysDestroy()  from Singleton destructor, and call it in another function before Singleton destructor.

another question ： when occur "paper jam" or "scan was called" and so on, popup alert message box, which cann't remove, and I cann't get the notification from TwainCallbackProc64, but after I set DTWAIN_SetErrorCallback, I can get above error from error  callback after popup dismiss; is there a function to get above error before popup message box showing? or Is there any way to known the status and avoid these alert boxer popup

dynarithmic, 2024-11-28:
> I had solved the crash problem, just remove DTWAIN_SysDestroy() from Singleton destructor, and call it in another function 

Depending on how you wrote the singleton, you cannot rely on calling cleanup functions in the destructor of a singleton if you do not have control of when the singleton is destroyed.  This is probably why the DSM crashed, and that is because all of the infrastructure required by TWAIN to properly close itself down was either destroyed, or in an inconsistent state.

For example, singletons that are global (or static) objects that only get destroyed by the runtime when the application terminates -- these should not be used if you want controlled destruction of the resources such as TWAIN.

Singletons that require the programmer to shut them down manually (i.e. dynamically created ones using *new* or similar, where you must call *delete*) are the only ones that are guaranteed to work, since you have full control of when the destructions take place.  

---- 

As to your second question:
DTWAIN cannot control how or when the source will report errors, as there is no direct control of how the Source's GUI operates.  

Assuming you are using Visual Studio -- if you were to run your application under the debugger, when the popup occurs you go to "Break All" and inspect the call stack, you will see the functions that led to the popup occurring.   More than likely you will see that the list of functions in the call stack all point to functions within the TWAIN driver itself, thus DTWAIN has no control over when or how that popup occurs.

As mentioned in [this closed question](https://github.com/dynarithmic/twain_library/issues/131#issuecomment-2493866277), there are custom capabilities available in the scanner, but you have to get the information from the manufacturer to determine what these capabilities do.

As to paper jams, if the device supports [device events](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/SpecialDeviceEvents.html), and in addition, support the notification for paper jams [DTWAIN_DE_PAPERJAM](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetDeviceNotifications.html), then a paper jam can be supported.  But again, the GUI has total control of what pop-ups occur.

leetb-leetb, 2024-11-29:
thanks dynarithmic，close the issue.

closed 2024-11-29 by leetb-leetb