dynarithmic/twain_library #134: Use Library without UI

jeepMan79, 2024-12-05:
Hi!

I would like to use the TWAIN library and be able to scan programmatically.  My goal is to encapsulate the scanning logic behind HTTP APIs to allow a Web Application to scan documents.  When I try to acquire the image, it throws an error saying "Cannot start UI".  Is what I'm trying to do possible?

Thanks!!

dynarithmic, 2024-12-06:
You will need to provide more information.  

> it throws an error 

1) If your scanner allows the scanner to operate without a user interface, then it can be used in this manner, however the error you are stating is not one that DTWAIN generates.  DTWAIN does not generate the error "Cannot start UI".  What component generates this error?  Is it your scanner that generates the error?

2) DTWAIN is simply a DLL that can be called from any environment that can call exported DLL functions.  What computer language are you using?

jeepMan79, 2024-12-06:
this can be closed.  Sorry, I was using the llibrary incorrectly.  Its working now.  Thanks!!

closed 2024-12-06 by jeepMan79