dynarithmic/twain_library #142: DTWAIN Demo Program shows source but immediately indicates device not found (HP Scanjet 7000 s3)

apt-rgn, 2025-02-18:
@dynarithmic 

Hi.  I am a coworker of sean-neeley, whom you have assisted in the past with issues like below.  
https://github.com/dynarithmic/twain_library/issues/119

We appreciate your past help and are hoping you can assist on this new issue.  I am new to this, so please excuse me if this post is too general.

As sean-neeley mentioned, our company has a document imaging product used to capture scanned documents with barcodes and link an image to the relevant transaction in our ERP software.   For example, a scanned signed invoice for a sales order.  The latest version of this product requires a TWAIN driver.

All of our clients using this product began experiencing issues with the Win 11 24H2 update.   To resolve, most rolled back the update and were able to continue using our product.  However, some continued to experience when rolling back and are having issues using our product.  We were able to replicate the issue in our demo system on a Win 11 22H2 machine.

I ran your DTWAIN Demo Program (64 bit) and experienced the same issue(s) as our product.

- We are able to select the source.  See 1st attachment.
- Then, we get a device not found error.  See 2nd attachment.
- Following,  we see a message regarding being unable to open the source.  See 3rd attachment.

[image]

[image]

[image]

I ran the DTWAIN Demo with logging enabled.  See 4th attachment.

[TWAINDSM.log](https://github.com/user-attachments/files/18855474/TWAINDSM.log)

At the beginning of the log, I see the following.  This file exists on my device.
[170852478 apps.cpp 1648   203 0000000000005764]   Loaded library: C:\WINDOWS\twain_64\HP Scan\hpxsTwain.ds (TWAINDSM_USEAPPID:1)
[170853836 dsm.cpp  1264     2 0000000000005764]   MSG_OPENDS failed...

Any advice you could provide would be appreciated again.

Thanks

dynarithmic, 2025-02-19:
I am running with the following:

```
Edition	Windows 11 Pro
Version	23H2
OS build	22631.4751
```

 Running the HP-7003 s3 for 64 bit, here is what I discovered:

1) DTWDEMO64U.exe does not open the source (selecting the source is ok).  The error looks like the same error you are getting.  I am selecting the "HP TWAIN USB" device.

2) For a test that doesn't use DTWAIN, I tried the [twack_64.exe](https://github.com/dynarithmic/twain_library/tree/master/TwainSpecification) program.  This program is the one found at the [Twain Working Group](https://github.com/twain/twain-toolkit/tree/master) site.  

To use it, you should always select "Select Source..." from the main menu, and then "Acquire".  You will see the same error that you're encountering now with the HP driver.  To verify it is an issue, you could try any other scanner that shows up in the 64-bit Select dialog -- you should see that it works correctly.

3) For a third test that is independent from DTWAIN, I tried the [64-bit Irfanview](https://www.irfanview.com/64bit.htm) program to select a source and acquire an image.  Irfanview failed to open the source,

4) I looked at the **c:\Windows\twain_64\HP Scan** directory, which is the one I believe is installed for the HP driver.  This directory is basically the one that the TWAIN DSM finds when determining the source names to be placed in the "Select Source" dialog.  The file with the .ds extension is the one that the DSM is using.   The directory consists of just three files:

```
HPScanUI.dll
hpxsTwain.ds
ScanApp.ini

```

5) I tried the 32-bit driver, and it seems that HP has abandoned their "true" HP driver for the generic WIA driver.  I only see WIA selections for the HP driver.  Selecting the WIA driver for 32-bit works correctly, but you only get the generic dialog that all WIA devices use, and not the true "HP" user interface.  I vaguely remember that their older, true, non-WIA driver(s) did work, but that was for Windows 10.

If I had the source code to the HP driver, I would be able to determine why the source could not be opened, but unfortunately I do not have access to the code.

So the conclusion I have is that there is something missing or wrong with the 64-bit HP driver or installation.  If the Twack_64.exe program and other programs independent from DTWAIN fail to work with a basic "open source", then DTWAIN and probably every program using the scanner will have a problem.

dynarithmic, 2025-02-19:
I did some further debugging, and the following exception occurs when attempting to open the HP device:

```
TRACE: LocalSettingsPath is 'C:\ProgramData\HP\PDSMQ\PdsmqLocalSettings.xml'
'DTWDEMO64D.exe' (Win32): Loaded 'C:\Windows\System32\msxml3.dll'. 
TRACE: FactorySettingsPath is 'C:\Repos-Copy\Dynarithmic\twain_library_source\experimental-logging\source\binaries\64bit\PdsmqConfig.xml'
Exception thrown at 0x00007FFB6C21FB4C in DTWDEMO64D.exe: Microsoft C++ exception: HP::Pdsmq::Common::FileNotFoundException at memory location 0x0000007F4E976950.
ERROR: Fail creating Settings class: No factory settings file found
```

The most important piece of information is this:
**ERROR: Fail creating Settings class: No factory settings file found**

On my machine, there are no xml files on either the 
C:\ProgramData\HP\PDSMQ\ 

or the 

C:\Repos-Copy\Dynarithmic\twain_library_source\experimental-logging\source\binaries\64bit\ 

paths (the latter makes sense, because I would need to manually install this file at the location I am running the demo on).  

So it does seem like HP is forgetting something on installation, more specifically those xml files.   

The only thing I can do is to enhance the error reporting, since the exception error and reason is coming through to DTWAIN, but not reported.  Other than that, I suggest going to HP and resolving this issue, or try and get one of your customers to verify that they have these missing xml files.

dynarithmic, 2025-02-19:
I have a debug version of TWAINDSM.dll, which is the Data source manager for 64-bit devices.  Unfortunately the exception error is not thrown back to DTWAIN to process, and instead TWAINDSM.dll consumes it and carries on to deliver the return code of "error" back to DTWAIN (and any other application).

If you need to see the error yourself, you can debug your application and take a look at your debugger's "output" window (I am assuming Visual Studio, if this is available).  This will show all exceptions thrown by the DLL's that are unhandled -- you should see the one I mentioned in the previous comment.

As I suspected, it seems that the older, Windows 10 drivers running on Windows 10 works correctly for 64-bit applications.  I had an older Windows 10 machine with an HP driver installed, and indeed it installed those XML files (at least I see the **PdsmqLocalSettings.xml** file).  I copied that XML file over to the Windows 11 machine, and an exception is still thrown from the HP driver, so the XML file can't simply be copied from another machine -- it probably needs to be officially installed and registered in some way.  For the users that do have success running Windows 11, you should check if they have those XML files installed. 

As to the HP drivers, HP seems to have changed things up on their drivers, at least for Windows 11 -- no longer do I see non-WIA drivers, it seems that the 64-bit version installation simply is buggy or incomplete.

I would suggest that you send another program to your users  that uses TWAIN (something small) that is independent of your application.  The users that are still having issues should try to run this program.  This will, at the very least, convince them that it isn't just your application, but instead is a general HP problem that needs to be addressed by HP.  A reinstallation of the drivers didn't change the issue on my end.