dynarithmic/twain_library #143: Error code 1002 with Avision scanner

johandv, 2025-03-11:
Hello,

We've integrated an Avision IDA6 scanner into a Java8 32 bit application which is installed for +-1000 clients.
But for 1 client, everything is working fine everywhere.
We're trying to figure out why they are having troubles. (Everyone is using the same version of our software).

Most of the time scanning works perfectly fine for them, but at some point the image cannot be retrieved anymore from the scanner (we've tested both DTWAIN_AcquireFile and DTWAIN_AcquireNative). The return code we see is 1002 (which indicates acquire cancelled if I'm not mistaken). We don't show the UI while scanning, so the user can't cancel the scan manually.  In some cases they restart the scan just after that and it is working again , in some cases they receive the 1002 error again.

In attachement I've added the error-case and a success-case. Can you see any indication of what might cause this behaviour? 

Kind regards,
Johan

[0305Error.txt](https://github.com/user-attachments/files/19178904/0305Error.txt)
[0305Success.txt](https://github.com/user-attachments/files/19178905/0305Success.txt)

dynarithmic, 2025-03-11:
If you compare both files, The 0305Error.txt has a failure here (search for the word "FAILURE"):

`DSM_Entry(pSource=3D100634H, pDest=795DA39CH, DG_IMAGE, DAT_IMAGENATIVEXFER, MSG_GET, TW_MEMREF=4B95E570H) called
...
[2025-03-05 14:45:30] =1 (TWRC_FAILURE)`

If this call fails, then the image will not be acquired.  Note that DTWAIN really has no control over this, as this makes a call to the TWAIN system,  then the TWAIN system (actually, TWAIN_32.DLL or TWAINDSM.DLL) makes a call to the Avision driver to start the native transfer.  The triplet `DG_IMAGE / DAT_IMAGENATIVEXFER / MSG_GET` is the one that must return successfully from the driver for any acquisitions to take place.

Note that the 0305Success.txt doesn't have these failures, except for the expected ones when querying for some device capabilities.

----
First, I would suggest getting the latest version of DTWAIN.  The current version is 5.6.1, which is 3 (or more)  versions greater from what you are currently using.  In addition, the latest Java interface and JNI DLL's were updated in the later versions.

Second, there really isn't anything I can do specifically, since a) this is sporadic issue and b) I really don't have the device to test.  

Also but minor, your logs seem to be missing the text resource files (twainresourcestrings_english.txt, for example).  I see that some text is missing from the logs.

dynarithmic, 2025-03-11:
There is also these series of lines in the error log:

```
DTWAIN Message(HWND = 00000000, MSG = 49814, Notification code = 1038, LPARAM = -1101

[2025-03-05 14:45:30] 
DTWAIN Message(HWND = 0073051E, MSG = 49814, Notification code = DTWAIN_TN_TWAINPAGEFAILED, LPARAM = 2036179736

[2025-03-05 14:45:30] 
DTWAIN Message(HWND = 0073051E, MSG = 49814, Notification code = DTWAIN_TN_PAGEFAILED, LPARAM = 2036179736

```
Possibly the Java interface doesn't get these codes (I would need to look at the JNI source code), but definitely DTWAIN is sending the DTWAIN_TN_TWAINPAGEFAILED and DTWAIN_TN_PAGEFAILED notifications back to the application on failure.

johandv, 2025-03-12:
Thanks for your answer,

First of all we'll update to the 5.6.1 version and test with this version.

As only one client has this sporadic issue, do you think there is any chance the Avision-hardware itself might cause this instability? Or any missing software component on their computer? The Visual C++ installation looks fine to me according to the explanation.

Kind regards,
Johan

closed 2026-09-05 by dynarithmic