dynarithmic/twain_library #144: trouble getting started with Python 3 (Windows 11)

dmccallie, 2025-03-13:
Hello, this looks like a **very** useful library, so I hope you can help me get started using it with Python 3.10, under Windows 11. I have run into a few problems.

I am using the 64-bit version of the unicode DLL.
The DTWDEMO64U program works correctly to access a local Epson scanner (over local network).

This test program, modified from your python example, works well until I get to the `dtwain_dll.DTWAIN_GetSourceProductNameW()` call, which always fails with return value of -1 (DTWAIN_FAILURE1)

```python
import dtwain
import ctypes as ct

def test_dtwain():
    
    # load the 64-bit unicode version of the dtwain dll
    dtwain_dll = ct.windll.LoadLibrary(".\\release_libraries\\dtwain64u.dll") 
    
    # is twain available?
    if not dtwain_dll.DTWAIN_IsTwainAvailable():
        print("Twain is not available")
        return
    else:
        print("Twain is available")  # prints Twain is available

    # Initialize DTWAIN
    rv = dtwain_dll.DTWAIN_SysInitialize()
    print(f"DTWAIN_SysInitialize returns: {rv}")  # returns 1
    
    # set up logging
    log_file = ct.create_unicode_buffer(".\\dtwain_log.log")
    rv = dtwain_dll.DTWAIN_SetTwainLogW(dtwain.DTWAIN_LOG_ALL, log_file) 
    print(f"DTWAIN_SetTwainLogW returns: {rv}") # returns 1
    
    # are we initialized?
    rv = dtwain_dll.DTWAIN_IsInitialized()
    print(f"DTWAIN_IsInitialized returns: {rv}")  # returns 1
    
    # Select a TWAIN source
    TwainSource = dtwain_dll.DTWAIN_SelectSource() # returns a non-zero handle to the selected source
    if TwainSource:
        # Display the product name of the Source
        mystrbuf = ct.create_unicode_buffer(30)
        rv = dtwain_dll.DTWAIN_GetSourceProductNameW(TwainSource, mystrbuf, len(mystrbuf)) # returns -1
        if rv == dtwain.DTWAIN_FAILURE1:
            print(f"rv = {rv} Failed to get source product name")
        else:
            print(f"Get source returns: {rv} with Source Product Name: {mystrbuf.value}")

        # Acquire to a BMP file
        rv = dtwain_dll.DTWAIN_AcquireFile(TwainSource, ".\\TEST.BMP", dtwain.DTWAIN_BMP, dtwain.DTWAIN_USELONGNAME,
                                      dtwain.DTWAIN_PT_DEFAULT, 1, 1, 1, 0) # returns 0
        # rv = dtwain_dll.DTWAIN_AcquireNative(TwainSource, dtwain.DTWAIN_PT_RGB , 1, 1, 1, 0)
        print(f"DTWAIN_AcquireFile returns: {rv}")  # returns 0

    # Close down DTWAIN                                      
    rv = dtwain_dll.DTWAIN_SysDestroy()

if __name__ == '__main__':
    test_dtwain()
```

When I run the program, the source select window pops up and I can select my Epson printer, but when I query for the name of the `TwainSource`, I consistently get an empty string, with return code of -1.  Note that I am using Python3, which is always Unicode as best I understand, and so I am using the "W" variants of the DLL entry points, as well using `create_unicode_buffer()` instead of `create_string_buffer()`

I tried all combinations of using the "A" or "W" or "" versions and with `create_string_buffer()` but that made no difference.

When I look at the logfile, I can see the Epson printer name, so I know that I am connecting to the actual device.

The call to `dtwain_dll.DTWAIN_AcquireFile()` also fails.

[dtwain_log.log](https://github.com/user-attachments/files/19221457/dtwain_log.log)

The logfile is attached, if that might help.

I suspect I am doing something stupid, but I can't find it!

Thanks in advance.
--david

dynarithmic, 2025-03-13:
The issue seems to be that the 64-bit version of ctypes is not returning pointers to 64-bit items.  

I took your code and ran it in 64-bits.  The source handle you are retrieving has truncated the upper 32-bits.  For example, if the source handle is 0x000001231a345678, ctypes is giving you 0x000000001a345678.   You can see this in the logs here:

```
[2025-03-12 22:25:12.913] Thread [37616] <<<===Exiting DTWAIN_SelectSource returning value of 000001E4E449CD90
[2025-03-12 22:25:12.913] Thread [37616] ===>>>Entering DTWAIN_GetSourceProductName(Source=0x00000000E449CD90, szProduct0x000001E4E2A357B0, nMaxLen=1e)

```
Note that the value that DTWAIN_SelectSource returns is `000001E4E449CD90`, but the value you are using in DTWAIN_GetSourceProductName is `00000000E449CD90`.  Note that the upper 32-bits are zeroed out.

To get more technical, the failure occurs when DTWAIN checks to see if the source handle you are using is valid.  Since the source handle is not valid, -1 is returned.  If you were to call DTWAIN_GetLastError(), you should get a return code that the source is invalid (DTWAIN_ERR_BAD_SOURCE).

I will be honest with you -- I am not a python expert with respect to ctypes.  For some reason, the 32-bit version of python works correctly due to pointers being 32-bit.   However the truncation issue for 64-bit is a bit out of my reach.  I always had suspicion about the 64-bit version of python and ctypes working properly.

So the issue is really ctypes and setting it up properly to handle 64-bit pointers being returned.  If you can get this to work correctly for 64-bit, I will be happy to amend the sample code (the sample does work for 32-bit).

dynarithmic, 2025-03-13:
I experimented a bit with this:

`dtwain_dll.DTWAIN_SelectSource.restype = ct.POINTER(ct.c_void_p)
`
and doing this allows DTWAIN_SelectSource to return the full 64-bit value.  This should now allow the calls to DTWAIN_GetSourceProductName() and DTWAIN_AcquireFile() to execute properly.

dynarithmic, 2025-03-13:
It looks like the python binding needs to add an entire function that sets up around 150 or so DTWAIN functions that returns pointers.  Either that or the programmer has to code in each line to define the return type as `ct.POINTER(ct.c_void_p)` depending on which functions the app will call.

The main functions that need to be added are all the ones that return the following:

```
DTWAIN_ARRAY
DTWAIN_FRAME
DTWAIN_HANDLE
DTWAIN_OCRENGINE
DTWAIN_RANGE
DTWAIN_SOURCE
HANDLE

```

dynarithmic, 2025-03-13:
Also, you should add the DTWAIN_LOG_USEFILE flag:

`rv = dtwain_dll.DTWAIN_SetTwainLogW(dtwain.DTWAIN_LOG_ALL | dtwain.DTWAIN_LOG_USEFILE, log_file) `

Without this flag, the current DLL version and other information is missing from the log.

dmccallie, 2025-03-13:
Wow!  Thank you for the **awesome** support!

I don't yet know which functions I will need, so I plan to wrap them with the proper ctypes decorations on an as-needed basis, at least for now.

Thanks again!

--david

dynarithmic, 2025-03-13:
Addressed issue in commit in the [twain_library_source / development](https://github.com/dynarithmic/twain_library_source/commit/f2554cc5b6da2e07a98c5ccad38df99f9fa27e33) branch.

dmccallie, 2025-03-13:
This looks great!

I am still struggling with how to properly setup and use the functions that use DTWAIN_ARRAY. In particular, I am confused on the difference between initializing for routines that expect `LPDTWAIN_ARRAY` versus those that expect `DTWAIN_ARRAY`

For example, I have not yet figured out the ctypes incantations to get `DTWAIN_EnumFileXferFormats()` and `DTWAIN_ArrayGetAtLong()` to work correctly.

I was able to get `DTWAIN_GetResolution()` to work correctly using:
```python
        rv = dtwain_dll.DTWAIN_GetResolution.argtypes = [ct.c_void_p, ct.POINTER(ct.c_double)]
        res = ct.c_double(0)
        rv = dtwain_dll.DTWAIN_GetResolution(TwainSource, ct.byref(res))
        print(f"DTWAIN_GetResolution returns: {rv} resolution: {res.value}")
```

but I have not been able to replicate that pattern for the DTWAIN_ARRAY routines.  I'll keep working on it, but I have to be away for a while.

Thanks again for you help!

--david

dynarithmic, 2025-03-14:
A DTWAIN_ARRAY is actually just a 64-bit pointer type (or 32-bit if the python app is 32-bit).  So for an LPDTWAIN_ARRAY you aren't passing the pointer, but the address of the pointer itself.

You can try this:

```
    # declare a pointer
    dtwain_array = ct.pointer(ct.c_void_p(0))

    # pass the address of the pointer (LPDTWAIN_ARRAY)
    dtwain_dll.DTWAIN_EnumFileXferFormats(TwainSource, ct.byref(dtwain_array))

```
Note that you will need to destroy the array when done.

`dtwain_dll.DTWAIN_ArrayDestroy(dtwain_array)`

dynarithmic, 2025-03-14:
Reopening issue.

If the issue with handling LPDTWAIN_ARRAY is resolved, you can close this issue out.

dmccallie, 2025-03-14:
Hello again,

The follow code seems to work but I am getting strange replies from the enumeration of transfer formats:

```python
        # enumerate the supported source file transfer modes
        # get the address of a pointer to the DTWAIN_ARRAY
        dtwain_array = ct.pointer(ct.c_void_p(0))

        dtwain_dll.DTWAIN_EnumFileXferFormats.argtypes = [ct.c_void_p, ct.c_void_p]
        dtwain_dll.DTWAIN_EnumFileXferFormats.restype = ct.c_bool
        
        rv = dtwain_dll.DTWAIN_EnumFileXferFormats(TwainSource, ct.byref(dtwain_array))
        print(f"DTWAIN_EnumFileXferFormats returns: {rv}") # returns True
        
        # access the elements of the array of dtwain_longs
        dtwain_dll.DTWAIN_ArrayGetAtLong.restype = ct.c_bool
        dtwain_dll.DTWAIN_ArrayGetAtLong.argtypes = [ct.c_void_p, ct.c_long, ct.POINTER(ct.c_long)]
        
        for i in range(10):
            long_val = ct.c_long(0)
            rv = dtwain_dll.DTWAIN_ArrayGetAtLong(dtwain_array, i, ct.byref(long_val)) # returns True for four elements
            if rv:
                print(f"DTWAIN_ArrayGetAtLong returns: {rv} with value: {long_val.value}") # returns 2,4,0,6

        dtwain_dll.DTWAIN_ArrayDestroy(dtwain_array)
```

In particular, the call to `EnumFIleXferFormats()` returns True, and the subsequent calls to `ArrayGetAtLong()` returns True for four entries, but the four entries are 2,4,0,6 which don't match FileXferFormats.  I assume I am doing something wrong?

Thanks
--david

dynarithmic, 2025-03-14:
In addition to this, if you plan to use TWFF_TIFF, you may want to call `DTWAIN_EnumCompressionTypes` to see if one of them can be used in conjunction with TIFF.  The compression types are defined in `twain.h`, and start with the name `TWCP_`.

dmccallie, 2025-03-14:
So helpful, thank you!

I have a related but more functional question - perhaps out of bounds here.  If so, that's OK.

I volunteer with a non-profit that does post-election ballot audits. We use a 100ppm scanner (older model Fujitsu, to be upgraded this year to a more recent model) to scan the front/back of the ballots into a directory, from whence we do our audits.  We process ballots in batches of ~250, using a ADF and duplex mode.  In the past, our code used SANE under Linux, but this is no longer supported, so we hope to port to Twain (hence all these questions!)

We scan at 300dpi in color, though gray would probably be fine. Each front/back becomes a separate numbered sequential JPG file.

In terms of the two main transfer options - USESOURCEMODE (file writes controlled by scanner driver?) vs [ USENATIVE or USEBUFFERED ] (file writes controlled by DTWAIN?),  which do you think would be best?  

Error management is our major concern. For example, if the scanner misfeeds, we'd like to have it stop so the operator can address the feed, and then proceed without having to reload the whole batch.  

(Ideally, we'd also like to perform a rapid image assessment (barcode, ballot orientation) before proceeding to scan the next page, but I'm not sure that's possible in ADF mode?)

It seems like for the control we want, we should use the DTWAIN transfer mode (USENATIVE) - do you think moving the (large) image data from scanner to driver to application would slow the scan rates too much?  Can the ADF be stopped and re-started by DTWAIN if we found a "bad" scan?

Sorry for so many questions, but you've been super helpful!

Thanks.

dynarithmic, 2025-03-14:
From the listing of the source's supported types, you have TWFF_JFIF, which is basically JPEG. 

Ok, just to recap, you have two choices for the file types:  either use DTWAIN_JPEG, or TWFF_JFIF with the USESOURCEMODE flag.   Whichever one turns out "better" in terms of speed, I am not sure.  

For DTWAIN_JPEG, the image comes in as a DIB (a BMP file without the header info) from the scanner, and then DTWAIN converts it (after possible resampling) to JPEG and then saves to the JPEG file.  The quality factor ([DTWAIN_SetJpegValues](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_SetJpegValues.html)) also has a hand in this -- the higher the factor, the more detailed the JPEG image.  

For USESOURCEMODE and TWFF_JFIF, the quality factor needs to be communicated to the device, and you achieve that by setting the ICAP_JPEGQUALITY capability (as described at the help file link above).  Now, how your device handles the entire process of producing the JPEG image -- I don't really know, since that is proprietary to the manufacturer -- maybe they have their own codec internally, or quite possibly the manufacturer farms out the JPEG creation to a third-party DLL (which wouldn't be surprising).  Regardless, the device should produce a JFIF / JPEG image at the end.

As to error management:  The only way to test this is by experimentation by forcing a paper jam to occur.  There *may* be a way to detect paper jams, using [notification processing and retries](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/HandlingPageAcquisitionFailures.html) and/or [DTWAIN_GetDeviceEvent](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/DTWAIN_GetDeviceEvent.html) (if the device supports this).  

But again, this takes experimentation -- sometimes the driver simply won't rescan if a jam occurs and just reports that the paper is jammed and takes manual intervention to open the scanner, and maybe restart the call to DTWAIN_AcquireFile() all over again.  It is also hit or miss with the ADF if the paper jams -- it all depends on the device.

As to barcode detection -- it is supported by DTWAIN if your scanner reports back to DTWAIN that [barcodes can be detected](https://www.dynarithmic.com/onlinehelp/dtwain/newversion/HandlingExtendedImageInformation.html).  For most scanners, barcode detection is done via third-party, and is not actually built into the scanner.  

Currently, the DTWDEMO program is being updated to detect barcodes when the image is being transferred and display the barcode information.  This won't be ready until DTWAIN version 5.6.2 is released.  

Right now, you can see most of the supported extended information (which barcodes are part of) for your scanner in JSON format by calling `DTWAIN_GetSourceDetails` (this is currently undocumented, but the prototype is:  `LONG DLLENTRY_DEF DTWAIN_GetSourceDetailsA(LPCSTR szSources, LPSTR szBuf, LONG nSize, LONG indentFactor, BOOL bRefresh)
`
 so you would call it something like this:

`DTWAIN_GetSourceDetailsA(source_product_name, character_buffer, character_buffer_size, 2, 1)`
 
Alternately, you can use the DTWDEMO program to display this information by selecting a source and viewing the Source Properties from the main "Source Selection Test" menu (the JSON information is on the rightmost pane of the dialog box).

dynarithmic, 2025-03-14:
Just to let you know, I have a Fujitsu fi-6130.  I haven't been able to get the barcode support working for it, since the driver doesn't seem to enable the detection even though the scanner reports that it can detect barcodes (or I haven't been able to find a sample page that can be fed into the scanner and have it detect the barcodes ).  Maybe you will get a better model of Fujitsu than the one I currently have.

I also have an Avision 176U, and the 32-bit driver's barcode support is there and working.  The 64-bit version has issues (it is an older scanner, and it seems Avision didn't put the effort into their later 64-bit driver as they did with their 32-bit driver).

dmccallie, 2025-03-14:
I have a simple Epson 3750 scanner/printer that I have been testing with.

It has an ADF, but when I used `DTWAIN_AcquireFileW()` to attempt to read 1 page from the ADF, it went ahead and read all the pages in the ADF.  Is that a typical experience, or is there something else I need to do to limit it to one page at a time from the ADF?

(My intent is to change to "native" mode but experimenting with file mode for now.)

The snippet of code used:
```python
   rv = dtwain_dll.DTWAIN_AcquireFileW(TwainSource, ".\\TEST-dwtainmode-noui-100-adf.jpg",
           dtwain.DTWAIN_JPEG, # filetype for DTWAIN transfer
           dtwain.DTWAIN_USENATIVE + dtwain.DTWAIN_USELONGNAME + dtwain.DTWAIN_CREATEDIRECTORY, # file control flag
           dtwain.DTWAIN_PT_DEFAULT, # pixel type
           1, # num pages
           0, # bShowUI 
           1, # bCloseSource,
            0  # pStatus
           )
```

dynarithmic, 2025-03-15:
1) You will need to create a log file to see what happens when the number of pages is limited to 1.  

2) Some devices ignore the attempt to set the page count, and instead will feed all the pages.  If this happens, you could trap the DTWAIN_TN_QUERYPAGEDISCARD notification and "throw away" the images acquired after page N (you can see the DTWDEMO program do this when asked to "keep the image").

For 2) I have never set up callbacks for python (required for notification processing), so I can't really be of extensive help there.  Regardless, you may have to set up some sort of callback, since there will be a lot you will have to do while the scanning process is taking place.

dmccallie, 2025-03-16:
I gave up on the page count problem, but I have gotten python callbacks to work, so I am going to try MODELESS scanning to get more control.  I will close this and ask any new questions in a new thread.

Thanks again for excellent support.

closed 2025-03-16 by dmccallie